microsoft/Azure-Kinect-Sensor-SDK #744: Need documentation on merging multiple videos recorded using external synchronization having each sensor connected to it's own host PC

JayaSaiSontineni, 2019-09-09:
**Describe the bug**

**To Reproduce**

**Expected behavior**

**Logs**

**Screenshots**

**Desktop (please complete the following information):**
 - OS with Version: 
 - SDK Version:  
 - Firmware version:  

**Additional context**

JayaSaiSontineni, 2019-09-09:
I have connected two sensors to different Host PC's. recorded 10 seconds video and have received 2 .mkv files. 

How can I merge them as a single video?

tesych, 2019-09-10:
@JayaSaiSontineni, you can definitely merge them as a single video. Please check out [green screen sample](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/tree/develop/examples/green_screen) for guidence.

JayaSaiSontineni, 2019-09-17:
Hello @tesych , Can you please provide more examples if available for better understanding of the concept on how to acheive it.

tesych, 2019-09-23:
@JayaSaiSontineni , unfortunately we do not have more examples at the moment. Please check again soon.

closed 2019-09-23 by tesych

madelinegannon, 2020-07-30:
@tesych is there a reference or diagram for how to composite live data streaming from multiple host pcs? The [docs](https://docs.microsoft.com/en-us/azure/kinect-dk/multi-camera-sync#host-computers) suggest it's doable, but I've only seen references to single PC with multiple USB Controllers in practice (like #971).