microsoft/Azure-Kinect-Sensor-SDK #745: What is the power connector

wes-b, 2019-09-09:
I want to make my own power connector. What is the size and type of connector used to power the Azure Kinect?

wes-b, 2019-09-09:
The connector is a 4.5mm OD with 3.0mm ID and a pin diameter of 0.6mm. Similar to this connector: https://www.alibaba.com/product-detail/4506-DC-plug-4-5mm-0_60742810909.html?spm=a2700.7724857.normalList.20.61e761fcUIgZxg

closed 2019-09-09 by wes-b

greatwallet, 2021-03-08:
Is there any recommended amperage?

lukemc, 2022-02-03:
I did get those Alibaba cores out of desperation and made a few by casting sugru around the barrel but it wasn't ideal.

I did find another product (from a 'HP Notebook') that I could butcher and ended up just splicing off the useful part and remaking the cable using these  instead: https://www.amazon.co.uk/gp/product/B08K3K6LKR/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1