microsoft/Azure-Kinect-Sensor-SDK #747: Functionality of IsAvailableChanged() for A4K (C#)

bastiankayser, 2019-09-10:
Please provide a method to recognize if and when the Azure Kinect device becomes unavailable (disconnected, paused, stopped) similar to the IsAvailableChanged-Event in for the Kinect V2.
If the event structure does not fit the current pattern please provide some other mechanism to check the status of the Azure Kinect device. Thanks!

BlueprintBen, 2019-11-02:
I'll add that the index-based access of the devices combined with exclusive opening for even serial num reading is another factor limiting our ability to deal with device connections/hotplugging. The events will certainly help, though I expect this will only get us partway or produce a messier structure than a more robust enumeration/opening interface.

wes-b, 2019-11-04:
It would be great if you could open a request at https://aka.ms/azurekinectfeedback where you use the feedback from the community to prioritize the features users want the most.

Brent-A, 2019-12-06:
This is the C# equivilant of this native SDK support: #175

closed 2019-12-06 by Brent-A