microsoft/Azure-Kinect-Sensor-SDK #750: Visual Studio's error Code LNK2019: Unresolved external symbol

cmarrs, 2019-09-10:
When following along with with Microsoft Azure quickstart tutorial on building a Kinect body tracking application, we receive an error saying there are various unresolved external symbols.  A screenshot of some of the errors is shown below.   

[image]

The tutorial was copied exactly as listed on the website, and the corresponding NuGet packages were downloaded to our project files.

Tutorial Site:
https://docs.microsoft.com/en-us/azure/Kinect-dk/build-first-body-app

Versions:
Visual Studios 2015, Target Platform Version: 10.0.14393.0
Microsoft.Azure.Kinect.BodyTracking v0.9.2
Microsoft.Azure.Kinect.BodyTracking.Dependencies v0.9.0
Microsoft.Azure.Kinect.BodyTracking.Dependencies.cuDNN v0.9.0
Microsoft.Azure.Kinect.Sensor v1.2.0

soocheolyoon, 2019-09-10:
Did you check configuration and platform (debug, x64) ?

yijiew, 2019-09-16:
@cmarrs is the problem resolved after you change the project platform to be x64?

cmarrs, 2019-09-19:
Yes changing the platform to x64 worked, thank you!

yijiew, 2019-09-19:
Awesome! Thanks for confirming :)

closed 2019-09-19 by yijiew

sealfoss, 2020-11-15:
Hi, I'm having this exact same problem. 
However, I have VS Solution Configuration set to Debug and Solution Platforms set to x64 already. I have installed the Microsoft.Azure.Kinect.Sensor NuGet package per instructions in the Azure Kinect documentation. I've added the \sdk\include\ folders for both Azure Kinect SDK v1.4.1 and Azure Kinect Body Tracking SDK to PropertyPages >> C/C++ >> General >> Additional Include Directories. I've added the \tools\ folders for the same to Property Pages >> Linker >> General >> Additional Library Directories. And yet, when I build, the linker throws a LNK2019 error for every SDK call I make. 

What is also weird is that when I try to build provided example projects, that works no problem. 

Anyone have any idea what I've done wrong? I have to believe I've just mis-configured the project somehow. My ultimate goal here is to build a .dll Unity3D plugin. Reading and writing data to and from Unity so far hasn't been any problem. Talking to the Kinect SDK is my road block right now. Any help is appreciated. Thanks in advance.