microsoft/Azure-Kinect-Sensor-SDK #752: USB Driver for Azure Kinect Depth Camera not working

Maxim-Zhukov, 2019-09-10:
Azure Kinect DK connected via both embedded and external (Renesas) PCI-E USB 3.0 ports recognized by Windows only as Azure Kinect Microphone array.

USB Type-C - Type-A data cable is 1m length and 2.4A.

Azure Kinect is always flashing white, with AzureKinectFirmwareTool -l command is's found 1 device, but can't connect bacause of LIBUSB problem.

Microphone works fine.

In device manager there no cameras.

wes-b, 2019-09-10:
Flashing white means the device thinks it is connected to USB2.0 host controller. Double check your cable.

- Try another PC if you have one available. It will help confirm if you are having a USB port issue or a cable/device issue.
- In Device Manager the Renesas host controller entry should have `(Microsoft)` appended to the end to indicate the Microsoft driver is loaded.

Maxim-Zhukov, 2019-09-11:
Thanks for your reply.

I've tested Azure Kinect with 3 different computers, all of them work with Kinect One, it means that USB 3.0 ports are fine.
Renesas host controller marks as Microsoft, it's ok.
I've checked two UCB Type-C - Type-A cables, it charging mobile phones.

In all of cases Azure Kinect shows flashing white indication.

wes-b, 2019-09-11:
I think you have a bad USB cable. If the devices connects on 3 PC's and you get the same state in Device Manager for each then the issue is not likely the host controller. Charging over the cable though is not enough to eliminate the cable of having USB3 issues.