microsoft/Azure-Kinect-Sensor-SDK #754: NullReferenceException in LargeArrayPool.cs line 36

bastiankayser, 2019-09-11:
**Describe the bug**
```
System.NullReferenceException
  HResult=0x80004003
  Message=Object reference not set to an instance of an object.
  Source=Microsoft.Azure.Kinect.Sensor
  StackTrace:
   at Microsoft.Azure.Kinect.Sensor.LargeArrayPool.Rent(Int32 minimumLength) in D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\csharp\SDK\LargeArrayPool.cs:line 36
   at Microsoft.Azure.Kinect.Sensor.Allocator.AllocateFunction(Int32 size, IntPtr& context) in D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\csharp\SDK\Allocator.cs:line 317
```

**To Reproduce**
I get depth images from the sensor and use a transform to create the point cloud from the depth image. The code looks like the following:
```
using (Capture capture = AzureKinect.GetCapture())
{
	var targetArray = new Int16[depthImage.HeightPixels * depthImage.WidthPixels*3];
	var pointCloudImage = transform.DepthImageToPointCloud(capture.Depth, CalibrationDeviceType.Depth);
	
	unsafe
	{
		using (var pin = pointCloudImage.Memory.Pin())
		{
			fixed (void* targetArrayPointer = &targetArray[0])
			{
				Buffer.MemoryCopy(pin.Pointer, targetArrayPointer, targetArray.Length*2,
					pointCloudImage.Size);
				
	
			}
	
		}
	
	}
	captureDepth.Dispose();
}
```
This works for some time but eventually throws the NullReferenceException.

**Expected behavior**
No NullReferenceException

**Desktop (please complete the following information):**
 - OS with Version: Windows 10 Pro Build 18363
 - SDK Version:  1.2.0 (nuget package
 - Firmware version:  
Current Firmware Versions:
  RGB camera firmware:      1.6.102
  Depth camera firmware:    1.6.75
  Depth config file:        6109.7
  Audio firmware:           1.6.14
  Build Config:             Production
  Certificate Type:         Microsoft

ahoink, 2019-10-29:
Happens to me too, but only on rare occasions. I'm using SDK version 1.3.0 from the nuget package. Camera firmware is all the same as the original post.

CameronVetter, 2019-11-14:
Same here, on rare occassion I end up in this code:

                    if (x.IsAlive)
                    {
                        buffer = (byte[])x.Target;
                        if (buffer.Length >= minimumLength)
                        {
                            _ = this.pool.Remove(x);
                            return buffer;
                        }
                    }

x.isAlive will be true for the if condition but by the time the buffer assignment happens isAlive is false and Target is null.  Seems like there is a thread safety issue here.

robharrisonuk, 2019-12-21:
I've just had this problem and it was due to the garbage collector being called on another thread after the x.IsAlive test and before buffer is assigned. My fix was to test if buffer is null before using it (which according to the see WeakReference docs is a valid test - also see https://stackoverflow.com/questions/1687045/thread-safety-of-weakreference)

PR #1006 (2020-01-14): Fix for issue #754. 
Fixed thread safety issue in LargePoolArray.cs. Garbage collector can be called on another thread after the x.IsAlive test and before buffer is assigned. This can lead to a null pointer exception as x.Target will be null after the garbage collector has freed it. 
(Refer to https://stackoverflow.com/questions/1687045/thread-safety-of-weakreference for more info.)

## Fixes #754

### Description of the changes:
Add a test to see if buffer is null before using it.

### Before submitting a Pull Request:
- [x] I reviewed [CONTRIBUTING.md](https://github.com/Microsoft/Azure-Kine [... cut]
closed 2020-01-14 by Brent-A