microsoft/Azure-Kinect-Sensor-SDK #759: Error LNK2019	unresolved external symbol _k4abt_tracker_create referenced in function

fyang064, 2019-09-12:
I have an issue on setting up the Body Tracking in Vistual Studio 2019, even if I have added the  additional dependencies k4abt.lib in the linker, same issue occurred all the time. I'm wondering if the directory of the library file cannot be used directly.

tesych, 2019-09-16:
@fyang064 , thank you for the question. Can you please provide more information about the SDK version and the platform. Thanks.

wes-b, 2019-09-16:
same as #750?

fyang064, 2019-09-16:
Yes, it’s same issue. I didn’t change the debug setting to x64.
Thank you so much for your help!

closed 2019-09-16 by yijiew

fyang064, 2019-09-16:
Hi Yijie,

Can you guys share the resources for the Body Tracking algorithm, like some
paper? I am curious about what kind of images are you using, just based on
the depth one?

Thank you so much for your help!

Best regards,
Fan