microsoft/Azure-Kinect-Sensor-SDK #765: SegmentColor C# Sample versus GreenScreen sample

JpEncausse, 2019-09-13:
Hello,
Thanks for all the job done !

The Kinect Sample C# section provide a **SegmentColor** project that works very well and is easy to install with NuGet (THANKS) compared to the SDK very hard to make it work (for a n00b)

**Is your feature request related to a problem? Please describe.**

I wondering how this **SegmentColor** sample is related to the **greenscreen** sample from the SDK ?

**Describe the solution you'd like**

- Can you keep the simple and easy to understand C# SegmentColor 
- Can you add an advanced version working like greenscreen with 2 kinects and more accurate depth ?
- Can you add in tool/ folder of the release of the SDK a compiled version of greenscreen ?

JpEncausse, 2019-10-14:
Since latest update (SDK 1.3) I got weird image results with SegmentColor sample
- I update the sample to get a black background
- It seems there is a pattern on the hand
- (it works with SDK viewer)

[image]

samoteph, 2019-11-04:
Hello !

I have the same problem as @JpEncausse describes.
All dependencies are up to date.
The SDKViewer works perfectly too.
After a PM with @jpEncausse, it seems that he has the problem again but that it depends on the machine he is testing.

wes-b, 2019-11-04:
This looks like https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/840.

ETowleMS, 2019-12-10:
The segment color sample is intended to show how you can use a single camera to display only information within a certain depth range. 

The green screen sample expands on this concept by demonstrating how you can calibrate 2 cameras, capture a scene, and then display only the changes to that scene within a defined depth distance.

closed 2019-12-10 by wes-b