microsoft/Azure-Kinect-Sensor-SDK #767: 3D registration application

Sara1702, 2019-09-16:
Hey there,
I was wondering if there is an application available for the Azure Kinect DK for 3D registration such as the "3D scan" app available for the Kinect for XBox?
Does anyone has experience with such an application for the Azure Kinect DK? 
I am working on a project where we would like to register the 3D features of bones and we are currently considering if we want/can use this sensor for our purposes or not.

**Describe alternatives you've considered**
I looked up several applications, but none seem to be compatible with the new Azure Kinect DK and/or are available for Windows 10. An alternative would be to code an application on our own. However, this would probably take too much time since it is not the main focus of our project.

tesych, 2019-09-17:
@Sara1702 , thank you for your question! Unfortunately, we do not have such an application available for Azure Kinect DK. Since it is a dev kit, it is built for our development community to build those applications to solve their specific business problems.

closed 2019-09-17 by tesych

kamalaVJ, 2020-02-11:
@Sara1702 I am currently working on a similar problem where I need the 3D scan/model of the object. As you mention 3D scan doesn't seem to be compatible with teh Azure Kinect DK. I have also tried Skanect ( popularly used with Xbox Kinect) but that doesnt recognize the device either.
Were you able to achieve this through a different app/method?

Chris45215, 2021-03-18:
Considering that Microsoft does have code and examples which do this, but for the HoloLens/AR libraries rather than the Azure Kinect libraries (https://docs.microsoft.com/en-us/windows/mixed-reality/develop/platform-capabilities-and-apis/qr-code-tracking), is there any compatibility with the Azure Kinect?

I had hoped that the Azure Kinect would be compatible with the hololens, and the libraries would provide some assistance for 3D reconstruction or limb tracking beyond the Hololens' vision. But I don't see much (if any) integration between the platforms.