microsoft/Azure-Kinect-Sensor-SDK #774: How to test c# application (unit/integration, mock, pre-recorded playback)

StummeJ, 2019-09-20:
We're trying to mock (Moq) out the Kinect device, but have noticed that it's written in such a way to not allow this.  The end goal is to test that our application sets up the device correctly and processes an image capture correctly.

Is there any guidance on how to test an application that uses the Sensor SDK?

Cheers

MsDerekMa, 2019-09-24:
Thanks for the feedback. I don't have much familiarity with Moq. Do you have a sample of something you can share where you were trying to mock and were having issues with?

StummeJ, 2019-09-30:
I don't have a sample but we were looking to test configuration, and capturing a frame to convert to a depth point cloud.  Really any guidance on how to test with the Kinect not in the loop would be amazing.

tesych, 2019-09-30:
@StummeJ , we are working on C# record and playback and hopefully it will help to resolve the issue.

closed 2019-09-30 by tesych