microsoft/Azure-Kinect-Sensor-SDK #775: Add Azure Kinect Sensor SDK port to Vcpkg

UnaNancyOwen, 2019-09-20:
This is a suggestion that add Azure Kinect Sensor SDK to Vcpkg.
What do you think?

**Is your feature request related to a problem? Please describe.**

Vcpkg is useful C++ library management tool.
It will be helpful for many developers, if Azure Kinect Sensor SDK can be installed using Vcpkg.

**Describe the solution you'd like**

Add Azure Kinect Sensor SDK port to Vcpkg.
```
# I hope it looks like this command
vcpkg install k4a:x64-windows
```

tesych, 2019-09-23:
HI @UnaNancyOwen and thank you for your continues contribution! Can you share more information what is the value to add Vcpkg?

UnaNancyOwen, 2019-09-24:
Hi @tesych, I think it makes following values.

* Developers can easily set up Azure Kinect SDK with the same command on any platform (Windows, Linux) by using Vcpkg. It is same for other libraries (e.g. OpenCV) that will be often used with Azure Kinect.
  ```
  ./vcpkg install k4a opencv
  ```

* It would be difficult for beginners, that to setting properly configurations for Azure Kinect SDK in Visual C++. (e.g. include directories, library directories, additional dependencies, ...) Vcpkg can strongly support it for beginners. Just run following command.
  ```
  ./vcpkg integrate install
  ```

* It is useful when setting-up Azure Kinect SDK in CI environment, because it works with commands.

UnaNancyOwen, 2020-01-07:
Azure Kinect Sensor SDK port is merged in Vcpkg. Thanks @JackBoosY!
https://github.com/microsoft/vcpkg/pull/8786

closed 2020-01-07 by UnaNancyOwen