microsoft/Azure-Kinect-Sensor-SDK #782: Simple way to export grayscale depthmap & a Scanning App please

dirrogate, 2019-09-22:
As in the title - I think Microsoft needs to think of the AzureK being used by non coders.
The key to mass adoption (and innovation) is when the thinking behind the code being released has a way of reaching a wider demographic - I speak from a creative-filmmaker angle.

- There is an OpenCV example of using the AzureKinect to scan models/build an environment. Why cant MS release an .exe instead of making non coder have to jump through hoops to compile?

- How about a simple commandline/Gui based system to convert the 16bit depth images recorded by kinectrecorder (within the .mkv file) to a typical 8bit grayscale png sequence?

Hoping you consider this.
Regards.

tesych, 2019-09-30:
@dirrogate , thank you for your feedback. Azure Kinect is the dev kit, and therefore our focus is to build API for developers to work with the device so they can build their own applications. All code samples we released so far are for developers to learn how to use the code. At this point there's no plan to release complete applications for non developers.

closed 2019-09-30 by tesych

dirrogate, 2019-09-30:
That's just the point, @Tesych 
Would you rather have developers (read) coders working in a vacuum or would a healthy cross-developmental effort for the AzureK lead to exponential adoption of 'developed ideas' and create a feedback based loop.

Developers aren't necessarily (and don't need to be) creative / artistes. But they do need creative ideas to feed off of.

tesych, 2019-10-01:
@dirrogate we really appreciate your feedback! I would suggest to describe your idea on our [customer voice](https://feedback.azure.com/forums/920053) and if it will receive high number of votes, we will consider implementing it.

dirrogate, 2019-10-01:
Honestly, @tesych,  "Implement?" Isn't it a matter of compiling the source code that is already up on this Git and putting out an .exe like the AzureKinect recorder .exe?

I'm sorry, (just being politically correct using that phrase) but all that's being asked is the code be compiled into a functional .exe for developers out there that are using the AzureKinect. 
Since when did being a developer= mandatory coding prowess? 

I don't think code savvy developers are "Full Stack" either.  In this sense, them having creative knowledge to film volumetrically, find best use cases for Volumentic documenting of archeology / architecture etc...

Sorry (again) but I think it's myopic of a dept. at AzureKinect to think of developer kits as being solely for those writing code.
When the Oculus DK1 was launched - as a filmmaker, I was one of the first to put out content without touching a single line of code in Unity. Ideas and output put forth from the completion of that stereoscopic VR film gave birth to actual code being written for Adobe's immersive plugins.

It's a simple request really that will benefit sales of the AzureKinect. It's not my job per-se to accelerate sales. I can show you how, though. 
I can , and I guess I will have to, get it done by a third party - but that won't benefit the community as whole. 

Someone at AzureKinect Dev dept - needs to talk to alert Customer voice, and complete this loop.
Best Regards.