microsoft/Azure-Kinect-Sensor-SDK #783: Azure kincet view can't start Depth Camera

ThangUnity, 2019-09-23:
[image]
[image]
Please help me

wes-b, 2019-09-23:
I think you have a bad device. There is no reason you should be getting a TransactionId of 0x5555555. Do you have a 2nd Azure Kinect?

ThangUnity, 2019-09-24:
I have only a brand new Azure Kinect. why bad device? . What can I do?

wes-b, 2019-09-24:
Use the k4aviewer and see if you can start just the color camera. Your log indicates reading the depth calibration is returning an invalid transaction ID, and there is no reason for that except the device or firmware are broken.

You can also try a factory reset, and then restore firmware to the version you have above.
https://support.microsoft.com/en-us/help/4494277/reset-azure-kinect-dk

closed 2019-10-09 by wes-b