microsoft/Azure-Kinect-Sensor-SDK #784: RGB Camera has improper exposure control

msamples, 2019-09-23:
Tested on K4A 1.2, Ubuntu 18.04

Problem reproduces with the demo app "k4afastcapture_streaming", but also when lower-level APIs are used.

Ultimately k4afastcapture_streaming makes use of uvc_set_exposure_abs in uvc_camerareader.cpp

By putting different values into the manual exposure control, I can make the camera return images with exposures = {500us, 2500us, 66670us}, but nothing else.

I would expect to have finer granularity in the control of the camera exposure.

This may be an issue with the FW on the device.

wes-b, 2019-09-23:
Thanks @msamples, please see issue #647 for a definition of how exposure works, and please see #448 for an issue that impacts the actual setting.

wes-b, 2019-09-24:
https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/792

closed 2019-10-09 by wes-b