microsoft/Azure-Kinect-Sensor-SDK #788: How to read all the imu samples after each image capture?

nonlinear1, 2019-09-24:
Recently, I am busy add imu support to my rgbdslam. I want access all the imu samples between two successive image captures. I have written a small example about imu data access according to https://docs.microsoft.com/en-us/azure/Kinect-dk/retrieve-imu-samples . the doc give the sentence "The API also has sufficient internal queuing to allow you to only check for samples after each image capture is returned." and "To retrieve all the currently queued IMU samples, you can call k4a_device_get_imu_sample() with a timeout_in_ms of 0 in a loop until the function returns K4A_WAIT_RESULT_TIMEOUT. K4A_WAIT_RESULT_TIMEOUT indicates that there are no queued samples and none have arrived in the timeout specified." It means that I can capture all the imu datas after a image capture, that is the imu data is captured from the time of image capture to imu data access?
   I have tested the way of the retrieve-imu-samples of Kinect-dk . It doesn't give the imu data frome the time of  image capture to the time of imu data access, insteadly give all the imu data and cann't stop. 
the following is my codes of my imu data access example. Could someone know how to access the imu data between the two image capture?
#include <iostream>
#include <k4a/k4a.h>
#include <opencv2/core/core.hpp>
#include <opencv2/highgui/highgui.hpp>
#include <opencv2/imgproc/imgproc.hpp>
#include <stdio.h>
using namespace std;
using namespace cv;

template<typename T> Mat create_mat_from_buffer(T *data, int width, int height, int channels = 1)
{
    Mat mat(height, width, CV_MAKETYPE(DataType<T>::type, channels));
    memcpy(mat.data, data, width * height * channels * sizeof(T));
    return mat;
}

static string get_serial(k4a_device_t device)
{
    size_t serial_number_length = 0;

    if (K4A_BUFFER_RESULT_TOO_SMALL != k4a_device_get_serialnum(device, NULL, &serial_number_length))
    {
        cout << "Failed to get serial number length" << endl;
        k4a_device_close(device);
        exit(-1);
    }

    char *serial_number = new (std::nothrow) char[serial_number_length];
    if (serial_number == NULL)
    {
        cout << "Failed to allocate memory for serial number (" << serial_number_length << " bytes)" << endl;
        k4a_device_close(device);
        exit(-1);
    }

    if (K4A_BUFFER_RESULT_SUCCEEDED != k4a_device_get_serialnum(device, serial_number, &serial_number_length))
    {
        cout << "Failed to get serial number" << endl;
        delete[] serial_number;
        serial_number = NULL;
        k4a_device_close(device);
        exit(-1);
    }

    string s(serial_number);
    delete[] serial_number;
    serial_number = NULL;
    return s;
}

int main(int argc, char **argv)
{
    cv::Mat imBGRA, imBGR, imD, imIr, im_transformed_color_image;
    cv::Mat im_transformed_depth_image;
   
    int returnCode = 1;
    k4a_device_t device = NULL;
    const int32_t TIMEOUT_IN_MS = 1000;
    k4a_transformation_t transformation = NULL;
    int captureFrameCount;
    k4a_capture_t capture = NULL;
    k4a_image_t depth_image = NULL;
    k4a_image_t color_image = NULL;
    k4a_image_t ir_image = NULL;
    k4a_image_t transformed_color_image = NULL;
    k4a_image_t transformed_depth_image = NULL;
    k4a_imu_sample_t imu_sample;
    uint32_t device_count = k4a_device_get_installed_count();
    bool continueornot = true;
    if (device_count == 0)
    {
        cout << "No K4A devices found" << endl;
        return 0;
    }

    if (K4A_RESULT_SUCCEEDED != k4a_device_open(K4A_DEVICE_DEFAULT, &device))
    {
        cout << "No K4A devices found" << endl;
        if (device != NULL)
        {
            k4a_device_close(device);
        }
    }
    k4a_device_configuration_t config = K4A_DEVICE_CONFIG_INIT_DISABLE_ALL;
    config.synchronized_images_only = true;
    config.color_format = K4A_IMAGE_FORMAT_COLOR_BGRA32;
    config.color_resolution = K4A_COLOR_RESOLUTION_720P;
    config.depth_mode = K4A_DEPTH_MODE_NFOV_UNBINNED;//640x576.
    config.camera_fps = K4A_FRAMES_PER_SECOND_30;
    k4a_calibration_t calibration;
    if (K4A_RESULT_SUCCEEDED !=
        k4a_device_get_calibration(device, config.depth_mode, config.color_resolution, &calibration))
    {
        cout << "Failed to get calibration" << endl;
        return 0;
    }
    const int width = calibration.depth_camera_calibration.resolution_width;
    const int height = calibration.depth_camera_calibration.resolution_height;
   
  
    transformation = k4a_transformation_create(&calibration);

    if (K4A_RESULT_SUCCEEDED != k4a_device_start_cameras(device, &config))
    {
        cout << "Failed to start cameras" << endl;
        return 0;
    }

    if (K4A_RESULT_SUCCEEDED != k4a_device_start_imu(device))
    {
        cout <<"Failed to start imu\n" << endl;
        return 0;
   }    

    for(int index = 0; continueornot; ++index )
    {
	    // Get a capture
	    switch (k4a_device_get_capture(device, &capture, TIMEOUT_IN_MS))
	    {
	    case K4A_WAIT_RESULT_SUCCEEDED:
		break;
	    case K4A_WAIT_RESULT_TIMEOUT:
		cout << "Timed out waiting for a capture" << endl;
		return 0;
	    case K4A_WAIT_RESULT_FAILED:
		 cout << "Failed to read a capture" << endl;
		return 0;
	    }

	     // Get a depth image
	    depth_image = k4a_capture_get_depth_image(capture);
	    if (depth_image == 0)
	    {
		cout <<"Failed to get depth image from capture" << endl;
		
	    }

	    int depth_image_width_pixels = k4a_image_get_width_pixels(depth_image);
	    int depth_image_height_pixels = k4a_image_get_height_pixels(depth_image);
	    imD = cv::Mat(depth_image_height_pixels, depth_image_width_pixels, CV_16UC1, (void*)k4a_image_get_buffer(depth_image));
           
            // Get a color image
	    color_image = k4a_capture_get_color_image(capture);
	    if (color_image == 0)
	    {
		cout << "Failed to get color image from capture" << endl;
		
	    }
	    int color_image_width_pixels = k4a_image_get_width_pixels(color_image);
	    int color_image_height_pixels = k4a_image_get_height_pixels(color_image);
	    imBGRA = cv::Mat(color_image_height_pixels, color_image_width_pixels, CV_8UC4, (void*)k4a_image_get_buffer(color_image));
            cvtColor(imBGRA, imBGR, CV_BGRA2BGR);

            // Get a ir image
            ir_image = k4a_capture_get_ir_image(capture);
            int ir_image_width_pixels = k4a_image_get_width_pixels(ir_image);
	    int ir_image_height_pixels = k4a_image_get_height_pixels(ir_image);   
            imIr = cv::Mat(ir_image_height_pixels, ir_image_width_pixels, CV_16UC1, (void*)k4a_image_get_buffer(ir_image));
          
  
           // get a group imu data between two captures????
           while(k4a_device_get_imu_sample(device, &imu_sample, TIMEOUT_IN_MS) != K4A_WAIT_RESULT_TIMEOUT)
           {

		 printf(" | Accelerometer temperature:%.2f x:%.4f y:%.4f z: %.4f\n",
		 imu_sample.temperature,
		 imu_sample.acc_sample.xyz.x,
		 imu_sample.acc_sample.xyz.y,
		 imu_sample.acc_sample.xyz.z);
                 std::cout << "the time stamp of Acc = " << imu_sample.acc_timestamp_usec << std::endl;
                 std::cout << "the time stamp of Gyro = " << imu_sample.gyro_timestamp_usec << std::endl; 

           }
           
          
          std::cout << "index = " << index << std::endl;
    
    }
    

}

wes-b, 2019-09-24:
The documentation is saying that there is enough buffering that the SDK will hold enough IMU samples that developers can safely read all the IMU data after each capture to get caught up and not drop data.

If you want to know all data between 2 frames then use the DEVICE timestamp of each image and compare that to the device timestamp of IMU. Stop reading when you IMU timestamp exceeds last image device timestamp. This will work fine at 30 & 15 FPS, not sure though about 5FPS

nonlinear1, 2019-09-24:
Thank you very much for your detailed explanation

closed 2019-10-02 by wes-b