microsoft/Azure-Kinect-Sensor-SDK #795: How to convert infrared image to color camera

guanming001, 2019-09-25:
Hi, 

Is it possible to convert infrared image to color camera space?

rabbitdaxi, 2019-09-25:
[k4a_transformation_depth_image_to_color_camera_custom()](https://microsoft.github.io/Azure-Kinect-Sensor-SDK/master/group___functions_gac00dd00e7612a86382e3d0a130f276bb.html#gac00dd00e7612a86382e3d0a130f276bb). It is possible, you can refer to above function, and you can pass in depth image as the transformation example does, then you can pass the IR image as the "custom image" as type of K4A_IMAGE_FORMAT_CUSTOM16. Then choose the interpolation type you like for the IR image (resolution change require some interpolation happening)

guanming001, 2019-09-26:
@rabbitdaxi  Thank you for the suggestion!

closed 2019-09-26 by guanming001

m5mattin, 2020-06-24:
Hi, 
Do you have code example. Every time i try i get a error "returned failure in transformation_depth_image_to_color_camera_custom() ". 
It will help me a lot. 
Thank you.

wes-b, 2020-06-24:
What is ported by the K4A log?

m5mattin, 2020-06-25:
**The error :** 
"[2020-06-25 08:36:18.831] [error] [t=29376] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/transformation/rgbz.c (88): transformation_compare_image_descriptors(). Unexpected image descriptor. Descriptor 1: width_pixels: 512, height_pixels: 512, stride_bytes: 1024, format: 5. Descriptor 2: width_pixels: 512, height_pixels: 512, stride_bytes: 512, format: 6. 
[2020-06-25 08:36:18.832] [error] [t=29376] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/transformation/rgbz.c (730): transformation_depth_image_to_color_camera_validate_parameters(). Unexpected custom image descriptor, see details above.
[2020-06-25 08:36:18.832] [error] [t=29376] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/transformation/transformation.c (710): transformation_depth_image_to_color_camera_validate_parameters( &transformation_context->calibration, &transformation_context->depth_camera_xy_tables, depth_image_data, depth_image_descriptor, custom_image_data, custom_image_descriptor, transformed_depth_image_data, transformed_depth_image_descriptor, transformed_custom_image_data, transformed_custom_image_descriptor) returned failure in transformation_depth_image_to_color_camera_custom()"

I don't understand why.

**My config :** 
           config.color_resolution = K4A_COLOR_RESOLUTION_720P;
           config.depth_mode = K4A_DEPTH_MODE_WFOV_2X2BINNED;  

**How i get images :** 
            k4a_image_t color_image = k4a_capture_get_color_image(capture);
            k4a_image_t color_image_aligned;
            k4a_image_t depth_image = k4a_capture_get_depth_image(capture);
            k4a_image_t depth_image_aligned;
            k4a_image_t ir_image;
            k4a_image_create(K4A_IMAGE_FORMAT_CUSTOM16,
                                        512,
                                        512,
                                        512 * (int)sizeof(uint16_t),
                                        &ir_image);
            
            ir_image = k4a_capture_get_ir_image(capture);
            k4a_image_t ir_image_aligned;
            
            k4a_image_create(K4A_IMAGE_FORMAT_COLOR_BGRA32,
                                                 512,
                                                 512,
                                                 512 * 4 * (int)sizeof(uint8_t),
                                                 &color_image_aligned);
            
            k4a_image_create(K4A_IMAGE_FORMAT_DEPTH16,
                                                 1280,
                                                 720,
                                                 1280 * (int)sizeof(uint16_t),
                                                 &depth_image_aligned);

            k4a_image_create(K4A_IMAGE_FORMAT_CUSTOM16,
                                                 1280,
                                                 720,
                                                 1280 * (int)sizeof(uint16_t),
                                                 &ir_image_aligned);

**Transformation :** 
k4a_result_t result_transf = k4a_transformation_depth_image_to_color_camera_custom(transformation, depth_image, ir_image, depth_image_aligned, ir_image_aligned, K4A_TRANSFORMATION_INTERPOLATION_TYPE_LINEAR, 0);

Do you understand what append ?

rabbitdaxi, 2020-06-25:
@m5mattin please check the documentation of the API. The k4a_transformation_depth_image_to_color_camera API requires both input and output custom images have the same custom format, e.g. in your case, K4A_IMAGE_FORMAT_CUSTOM16. You can refer to issue #1188 and you will spot the solution there. Basically, please make sure you also create a CUSTOM16 image for the ir_image (it seems redundant, but that is what the current API code does, and it checks the image format inside the code).