microsoft/Azure-Kinect-Sensor-SDK #796: Can I turn on the camera without using the GPU?

ruis1996, 2019-09-25:
I want to use the terminal to store the depth map remotely, but sdk seems to have to use the desktop environment to open the depth camera. Is there any solution?

RoseFlunder, 2019-09-25:
You could maybe try to setup an x-server-dummy like this:

https://techoverflow.net/2019/02/23/how-to-run-x-server-using-xserver-xorg-video-dummy-driver-on-ubuntu/

rabbitdaxi, 2019-09-26:
@ruis1996 thank you for the feedback, may I ask, the machine that you want to remote in (the one connect to the device), are talking about linux or windows? 
- if it is linux, you can try what @RoseFlunder  suggested, and please take a look this issue #562 
- if it is windows, you should be able to do that without problem

ruis1996, 2019-09-29:
@rabbitdaxi  @RoseFlunder 
Thank you for your reply.
My sys is Ubuntu18.04.3-mate.
I will try the methods you provided, I will return feedback later, thank you.

ruis1996, 2019-09-29:
@rabbitdaxi  @RoseFlunder 
I tried @rabbitdaxi ‘s method, but it can't work in my system. Is there any other way to solve it?

rabbitdaxi, 2019-09-30:
@ruis1996 , can you also try "export DISPLAY=:1" not sure whether it is helping, but it seems helps me if I do using a putty remote to a linux machine through SSH

RoseFlunder, 2019-10-09:
@rabbitdaxi 
Your method works me when I have a monitor connected to the remote PC to which I connect via SSH.

When I disconnect the monitor and restart the remote pc and connect via SSH again, exporting a DISPLAY environment variable is not sufficent because the X-Server isn't running and no graphics driver is loaded as far as I can see it.

I am searching for a solution to use the sdk (sensor & body tracking) on a headless (no monitor connected) ubuntu machine.

For the moment connecting via SSH and starting the application would be sufficent.
In the future the app should run as a systemd service that will be started automatically as a service when the machine starts.

sjdrc, 2020-01-03:
Can someone from Microsoft please give an update on this issue?

There are various bits of information scattered over a few issues, but there is no official or simple confirmed working solution for using a kinect from a headless linux environment, and myself and others have tried workarounds suggested but with no success. This use case is very common for research purposes.

palavrov, 2020-01-15:
I would like to join @sjdrc request - we need k4a working on headless linux environment - ideally on nVidia Jetson i.e. right now depth plugin is only for x86 and requires display and opengl 4.4 which completely prevent us to use the scanner in our products.

closed 2020-01-29 by wes-b

sjdrc, 2020-01-29:
Why has this been closed? There is no documentation on how to use the Kinect in a headless environment.

RoseFlunder, 2020-01-30:
Just to give some information about how I am working at the moment with my headless Ubuntu PC:
I am using automatic login at Ubuntu (which will start a X-Server Session and the NVidia driver starts too) together with a ssh server.
I haven't tried to autostart any app yet, but when I connect via SSH after the ubuntu PC has started, I just have to set the DISPLAY env variable with:
`export DISPLAY=:0`
Afterwards I can start the app that is using the kinect sensor sdk together with the body tracking sdk without an error.

So the only the thing that is missing for you would be way to start up an app after the auto login is complete.
Previously I tried my luck with systemd service, but I wasn't successfull with that.
Guess it is not designed to start applications that need a X-Server session + running NVidia Driver.

wes-b, 2020-01-30:
Thanks @RoseFlunder for the update.

This is close because issue is a duplicate of #810.