microsoft/Azure-Kinect-Sensor-SDK #800: Publish SDK 1.3.0

tesych, 2019-09-25:
1.3.0 betta version is released and ready to be used!
MSI Link: TBD will be posted soon
Nuget Link: [1.3.0-beta](https://www.nuget.org/packages/Microsoft.Azure.Kinect.Sensor/1.3.0-beta.0)
Deb package link: TBD will be posted soon

**_Summary of `[changes](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/CHANGELOG.md)`_**
*Multi cameras synchronization and capture example (green screen)
*New k4a_calibration_color_2d_to_depth_2d API
*Dynamically linked Windows VS dependencies
*Clear error messages
and more..

RoseFlunder, 2019-10-09:
Will there be a new body tracking sdk release that is compatible with sensor sdk 1.3 soon?
The current version of the debian ubuntu package _libk4abt0.9_ can't be used with the _libk4a1.3_ package because it has a hard dependency to _libk4a1.2._
Moreover its not possible to install multiple versions of the sensor sdk package because they use the same location for certain files (i.e. the depthengine).

JpEncausse, 2019-10-14:
Thanks !

Can you update "Azure-Kinect-Samples" with the SDK 1.3 ?
https://github.com/microsoft/Azure-Kinect-Samples/tree/master/build2019/csharp

wes-b, 2019-10-14:
> Can you update "Azure-Kinect-Samples" with the SDK 1.3 ?
> https://github.com/microsoft/Azure-Kinect-Samples/tree/master/build2019/csharp

The goal of the team is to ensure all samples in the SDK repo always build and run as expected. As as result our automation builds and validates those samples daily. 

The sample repo has a lower maintance bar. We commit samples here knowing they worked when checked in. We expect over time they will become stale and we are ok with that, as these are the samples that have dependencies on other projects that also chang with time. Feel free to submit a PR to help bring samples forward with recent SDK's

StevenButner, 2019-10-24:
> _1.3.0 version is released and ready to be used!_
> **MSI Link:** [1.3.0](https://download.microsoft.com/download/e/6/6/e66482b2-b6c1-4e34-bfee-95294163fc40/Azure%20Kinect%20SDK%201.3.0.exe)
> **Nuget Link:** [1.3.0](https://www.nuget.org/packages/Microsoft.Azure.Kinect.Sensor/1.3.0)
> **Deb package link:** TBD will be posted soon
> 

I am unable to find the new device firmware (version 2.6.104) that is supposed to be released with this 1.3.0 overall version.   I've used the MSI link given with #800 and I've looked in the folder **C:\Program Files\Azure Kinect SDK v1.3.0\tools\firmware** .... but the binary image that is found there is the same version (and, after bit-wise comparison, the same content) as the firmware that was released with SDK version 1.2.0.

Can you please tell me how to get the newest released firmware for my cameras?

RoseFlunder, 2019-10-24:
You could download the 1.4.0-alpha nuget package from here and just extract it using 7-Zip or similar:
https://www.nuget.org/packages/Microsoft.Azure.Kinect.Sensor/1.4.0-alpha.0
There you will find the new firmware in the firmware folder.

wes-b, 2019-10-24:
The firmware in 1.3.0 is the same as 1.2.0 and no update is needed.  The firmware in 1.4.0-alpha is good. No regressions have been found.

StevenButner, 2019-10-24:
My reason for wanting to try the 1.3.0 release was based on some firmware changes that were mentioned in #530 (by @wes-b).  I'm working on time stamping and multi-device synchronization, so it sounds like the recent firmware updates (in FW 1.6.104075014) might be helpful.  Unfortunately, my systems are Linux based but the recent releases are not packaged yet for Linux. That's why I tried to get the firmware via the MSI link. As I understand it, the Nuget method is not available on Linux systems.

I only need the firmware's .bin file but it seems that due to packaging issues, I'm not able to get it.  Is there any other way that I can get the new firmware?

RoseFlunder, 2019-10-24:
You can download the nuget package just in the browser and open/extract it there too to get the new firmware. Just follow the link I posted and you will see a download link on the right side.

StevenButner, 2019-10-24:
Thank you, @RoseFlunder ....that's exactly what I needed to learn.  Now I have the new firmware!

closed 2019-11-06 by wes-b