microsoft/Azure-Kinect-Sensor-SDK #807: Color <-> IMU extrinsic calibration incorrect.

helenol, 2019-09-28:
**Describe the bug**
The provided IMU <-> Color calibration in the SDK, which is published in the ROS driver on start-up and is otherwise accessed through `k4a_calibration_.extrinsics[K4A_CALIBRATION_TYPE_COLOR][K4A_CALIBRATION_TYPE_ACCEL]` is 180 degrees off from what I get if I calibrate the sensors myself using kalibr ( https://github.com/ethz-asl/kalibr ). This is reproducible across at least 2 production sensors. Furthermore both production sensors have the same IMU to color extrinsics which makes me suspect this value is off CAD and not calibrated, so could be a mistake in the axes of the IMU from the CAD.

I've attached the serial numbers, shipped calibration, and my kalibr calibration for 2 sensors. My calibration is also not the best (didn't have access to better calibration targets), but the rotation is 180 degrees different from what ships.

[sensor2.txt](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/files/3665194/sensor2.txt)
[sensor1.txt](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/files/3665195/sensor1.txt)

With the flashed calibration, visual-inertial odometry (tried VINS-mono and rovio) does not work; with my calibration, it works.

**To Reproduce**
1. Get IMU-color extrinsics from either ROS driver or SDK call
2. Do IMU-color calibration using kalibr https://github.com/ethz-asl/kalibr/wiki/camera-imu-calibration

**Expected behavior**
Calibrations should match (at least roughly).

**Logs**

**Screenshots**

**Desktop (please complete the following information):**
 - OS with Version: Ubuntu 18.04
 - SDK Version:  1.2.0
 - Firmware version: 
```
Firmware Build Config: Release                                                                                                         
Firmware Signature Type: MSFT signed                                                                                                   
Firmware Version RGB: 1.6.102                                                                                                          
Firmware Version Depth: 1.6.75                                                                                                         
Firmware Version Audio: 1.6.14                                                                                                         
```

**Additional context**

rabbitdaxi, 2019-09-30:
@helenol thanks for the feedback. Have you had a chance to take a look this page https://docs.microsoft.com/en-us/azure/Kinect-dk/coordinate-systems, is the coordinate system align with what your the other calibration method?

helenol, 2019-09-30:
@rabbitdaxi Hi again Andrew, thanks for the pointer!
Took a much closer look at the published coordinate frames. Canonically yes, ROS (and OpenCV and kalibr) conventions are right-handed, image -> 3D coordinates match, and IMU frame is right-handed however the IMU is physically mounted.
Looks like the culprit may be the ROS driver, which appears to be doing some conversions to the raw IMU data: https://github.com/microsoft/Azure_Kinect_ROS_Driver/blob/melodic/src/k4a_ros_device.cpp#L733-L767 but not applying the same conversions when publishing the extrinsics, rendering those extrinsics invalid. I can close this and make an issue on the ROS driver.

rabbitdaxi, 2019-09-30:
@helenol thank you for the quick response! sounds good.

helenol, 2019-10-02:
Now tracked here: https://github.com/microsoft/Azure_Kinect_ROS_Driver/issues/82 thanks to @skalldri! Thanks for the help everyone. Closing issue.

closed 2019-10-02 by helenol