microsoft/Azure-Kinect-Sensor-SDK #808: Question about persisting joint level data

shreyapandit, 2019-09-30:
Hi there,

Is there a way to record joint information using the recorder https://docs.microsoft.com/en-us/azure/kinect-dk/azure-kinect-recorder? 

It looks like recording depth and RGB is available, is there any other recorder/viewer available which allows joint data to be saved? 

Thanks so much!

qm13, 2019-10-02:
@shreyapandit we currently do not support saving the joint data. You can record the depth data and play it back through the body tracking SDK.

What is your use case for recording the joint pose data?

billpottle, 2019-10-05:
@shreyapandit - You can loop through skeleton.joints[counter].position for each frame and then do any analysis or save the resulting data to a file normally. Is this what you are asking? This is what my application does. The final result is just a JSON file with every position for every frame, saved for later comparison and analysis. They have good examples in the simple_3d_viewer and jump_analysis_sample. 

@qm13 - Our use case is in saving and transmitting lots of samples, joint pose data (if accurate) is a great compression method in that it captures the unique attributes of a movement while taking far less storage space than the full MKV files.

shreyapandit, 2019-10-06:
@qm13 The use case is a simple app where the user can view in real time, the location of all joints overlayed over the RGB input frames from the Kinect.

@billpottle Thanks! I will take a look :)

monajalal, 2019-10-11:
we tried loading the captured depth video into the body tracking sdk by calling the .exe file in tools folder of body tracking sdk and giving the path to the output.mkv. However, it did not work.

How should we exactly load the depth captures that are already saved into an .mkv file to the body tracking SDK?

RoseFlunder, 2019-10-11:
You have to compile a own software if you want to use/save the joint data.
I think I already linked this repository in another issue from you but here you go again:
https://github.com/microsoft/Azure-Kinect-Samples/tree/master/body-tracking-samples

There are multiple examples that you can take as a starting point.
These example projects can be opened and compiled with Visual Studio.

Fore more information on how to use the body tracking sdk in your own software check these docs:
https://docs.microsoft.com/en-us/azure/kinect-dk/build-first-body-app
https://docs.microsoft.com/en-us/azure/kinect-dk/get-body-tracking-results

closed 2019-10-28 by qm13