microsoft/Azure-Kinect-Sensor-SDK #809: Minimum Body Length for Body Tracking

guzza100, 2019-10-01:
What is the minimum body length that can be tracked?  It would be nice to be able to track the body of infants (3-9 months) for various research purposes.  I believe the predecessors had a minimum length of 1 meter.  Can that be brought down to 2 feet or 0.6 meters?

tobysharp-msft, 2019-10-29:
Hi @guzza100. That's interesting, can you give any background on the nature of your scenario that requires skeletal tracking for infants? e.g. are they lying on the floor, how they are moving, and where is the camera positioned? The tracking quality is not only about the size of the body but also these other variables. Thanks

guzza100, 2019-10-29:
Hello @tobysharp-msft .  Sure.  We are looking at tracking infant motions from overhead while they are lying on their back.  We are specifically looking at tracking their kicking motions.  Infant kicking motions can help with early detection of certain disabilities like cerebral palsy. 

I did a quick test with a NAO robot (about 2 feet tall) and the sensor could pick up the joints while the NAO laid still on its back.  However, it seemed to lose track when the robot raises its leg.

guzza100, 2019-10-30:
@tobysharp-msft Another possibility is placing sensors on the floor and detecting the infant from the left and right side (if body tracking can work at this viewpoint).

qm13, 2019-12-03:
@guzza100 we would like to get a recording that demonstrates the problems you are reporting. We require a signed data release agreement along with the recording. Could you provide me with contact details so I can provide you both the agreement and upload instructions?

closed 2020-01-24 by qm13

rejnok93, 2022-09-13:
Hello, I tried to use Azure Kinect Body Tracking SDK to capture some small babies (0.5 - 1 year old, none of them can't stand yet), tried many positions and many distances from camera, but k4abt library never saw any human body to show (output data empty). Does microsoft know about any use case of Azure Kinect DK to capture small babies? Is there any option better than coding new Body Tracking Library that supports small babies from scratch?
I can go and record an MKV file of a small baby or babies (with agreement) that demonstrate the problem discussed here if @guzza100 not provided any data yet? Is this still actual?

qm13, 2022-09-13:
@rejnok93 This is expected. The proportions of a human are very different to adults. Results for AKDK body tracking are unreliable for humans under 7 years old. We looked in the past at retraining the model for infants (as there are a small number of interesting infant applications). Unfortunately getting the necessary data is difficult and the ROI is poor.

guzza100, 2022-09-13:
I was able to get some skeletal tracking on infants between 3 and 6 months.  The key seemed to be putting them in a onesie with colors that had noticeable color contrast with the mat they were laying on.  Give that a try if you haven't already.

Daksitha, 2023-06-07:
Hello, I'm working on a similar project involving the tracking of infant movements, specifically between the ages of 3-9 months, and I've noticed some challenges with the Azure Kinect SDK. Unfortunately,  our initial testing with the SDK with Azure Kinect could not detect the infant at all except when we zoomed the camera toward the infant hiding the leg. We plan to capture the interaction of infants with body tracking information. Is there an option to capture only upper body disregarding the lower body, hoping that would improve the detection?

Additionally,  I have access to a sample dataset of infant interactions and I'm considering training my own model to improve tracking. Can you provide any guidance or best practices for this process? Specifically, I'm curious to know whether any changes have been made to the SDK since this issue was closed that could impact tracking of smaller bodies such as infants? Also, are there any specific data requirements or considerations I should keep in mind when creating the dataset for training the model?