microsoft/Azure-Kinect-Sensor-SDK #811: Ubuntu: Cameras Not Booting, Deluge of USB errors

soalive-smm, 2019-10-02:
**Describe the bug**
Cameras fail to boot, with lots of USB errors in the shell.

**To Reproduce**
0. We have 2 Kinect Azure sensors plugged into USB ports on the machine
1. ./bin/k4aviewer
2. See error:

```
foo@foo:~/SrcBar/build_FooBarDependencies/Azure-Kinect-Sensor-SDK/build/bin$ ./k4aviewer 
libusb: error [_get_usbfs_fd] libusb couldn't open USB device /dev/bus/usb/008/004: Permission denied
libusb: error [_get_usbfs_fd] libusb requires write access to USB device nodes.
[2019-10-02 11:56:57.735] [error] [t=4447] ../src/usbcommand/usbcommand.c (305): find_libusb_device(). Unable to open LIBUSB at index 0 
[2019-10-02 11:56:57.735] [error] [t=4447] ../src/depth_mcu/depth_mcu.c (68): usb_cmd_create(USB_DEVICE_DEPTH_PROCESSOR, device_index, NULL, &depthmcu->usb_cmd) returned failure in depthmcu_create()
[2019-10-02 11:56:57.735] [error] [t=4447] ../src/sdk/k4a.c (133): depthmcu_create(index, &device->depthmcu) returned failure in k4a_device_open()
libusb: error [_get_usbfs_fd] libusb couldn't open USB device /dev/bus/usb/004/004: Permission denied
libusb: error [_get_usbfs_fd] libusb requires write access to USB device nodes.
[2019-10-02 11:56:57.748] [error] [t=4447] ../src/usbcommand/usbcommand.c (305): find_libusb_device(). Unable to open LIBUSB at index 1 
[2019-10-02 11:56:57.749] [error] [t=4447] ../src/depth_mcu/depth_mcu.c (68): usb_cmd_create(USB_DEVICE_DEPTH_PROCESSOR, device_index, NULL, &depthmcu->usb_cmd) returned failure in depthmcu_create()
[2019-10-02 11:56:57.749] [error] [t=4447] ../src/sdk/k4a.c (133): depthmcu_create(index, &device->depthmcu) returned failure in k4a_device_open()
libusb: error [_get_usbfs_fd] libusb couldn't open USB device /dev/bus/usb/008/004: Permission denied
libusb: error [_get_usbfs_fd] libusb requires write access to USB device nodes.
[2019-10-02 11:57:00.628] [error] [t=4447] ../src/usbcommand/usbcommand.c (305): find_libusb_device(). Unable to open LIBUSB at index 0 
[2019-10-02 11:57:00.628] [error] [t=4447] ../src/depth_mcu/depth_mcu.c (68): usb_cmd_create(USB_DEVICE_DEPTH_PROCESSOR, device_index, NULL, &depthmcu->usb_cmd) returned failure in depthmcu_create()
[2019-10-02 11:57:00.628] [error] [t=4447] ../src/sdk/k4a.c (133): depthmcu_create(index, &device->depthmcu) returned failure in k4a_device_open()
libusb: error [_get_usbfs_fd] libusb couldn't open USB device /dev/bus/usb/004/004: Permission denied
libusb: error [_get_usbfs_fd] libusb requires write access to USB device nodes.
[2019-10-02 11:57:00.641] [error] [t=4447] ../src/usbcommand/usbcommand.c (305): find_libusb_device(). Unable to open LIBUSB at index 1 
[2019-10-02 11:57:00.641] [error] [t=4447] ../src/depth_mcu/depth_mcu.c (68): usb_cmd_create(USB_DEVICE_DEPTH_PROCESSOR, device_index, NULL, &depthmcu->usb_cmd) returned failure in depthmcu_create()
[2019-10-02 11:57:00.641] [error] [t=4447] ../src/sdk/k4a.c (133): depthmcu_create(index, &device->depthmcu) returned failure in k4a_device_open()
libusb: error [_get_usbfs_fd] libusb couldn't open USB device /dev/bus/usb/008/004: Permission denied
libusb: error [_get_usbfs_fd] libusb requires write access to USB device nodes.
[2019-10-02 11:57:01.693] [error] [t=4447] ../src/usbcommand/usbcommand.c (305): find_libusb_device(). Unable to open LIBUSB at index 0 
[2019-10-02 11:57:01.694] [error] [t=4447] ../src/depth_mcu/depth_mcu.c (68): usb_cmd_create(USB_DEVICE_DEPTH_PROCESSOR, device_index, NULL, &depthmcu->usb_cmd) returned failure in depthmcu_create()
[2019-10-02 11:57:01.694] [error] [t=4447] ../src/sdk/k4a.c (133): depthmcu_create(index, &device->depthmcu) returned failure in k4a_device_open()
libusb: error [_get_usbfs_fd] libusb couldn't open USB device /dev/bus/usb/004/004: Permission denied
libusb: error [_get_usbfs_fd] libusb requires write access to USB device nodes.
[2019-10-02 11:57:01.707] [error] [t=4447] ../src/usbcommand/usbcommand.c (305): find_libusb_device(). Unable to open LIBUSB at index 1 
[2019-10-02 11:57:01.707] [error] [t=4447] ../src/depth_mcu/depth_mcu.c (68): usb_cmd_create(USB_DEVICE_DEPTH_PROCESSOR, device_index, NULL, &depthmcu->usb_cmd) returned failure in depthmcu_create()
[2019-10-02 11:57:01.707] [error] [t=4447] ../src/sdk/k4a.c (133): depthmcu_create(index, &device->depthmcu) returned failure in k4a_device_open()
libusb: error [_get_usbfs_fd] libusb couldn't open USB device /dev/bus/usb/008/004: Permission denied
libusb: error [_get_usbfs_fd] libusb requires write access to USB device nodes.
[2019-10-02 11:57:02.075] [error] [t=4447] ../src/usbcommand/usbcommand.c (305): find_libusb_device(). Unable to open LIBUSB at index 0 
[2019-10-02 11:57:02.076] [error] [t=4447] ../src/depth_mcu/depth_mcu.c (68): usb_cmd_create(USB_DEVICE_DEPTH_PROCESSOR, device_index, NULL, &depthmcu->usb_cmd) returned failure in depthmcu_create()
[2019-10-02 11:57:02.076] [error] [t=4447] ../src/sdk/k4a.c (133): depthmcu_create(index, &device->depthmcu) returned failure in k4a_device_open()
libusb: error [_get_usbfs_fd] libusb couldn't open USB device /dev/bus/usb/004/004: Permission denied
libusb: error [_get_usbfs_fd] libusb requires write access to USB device nodes.
[2019-10-02 11:57:02.088] [error] [t=4447] ../src/usbcommand/usbcommand.c (305): find_libusb_device(). Unable to open LIBUSB at index 1 
[2019-10-02 11:57:02.088] [error] [t=4447] ../src/depth_mcu/depth_mcu.c (68): usb_cmd_create(USB_DEVICE_DEPTH_PROCESSOR, device_index, NULL, &depthmcu->usb_cmd) returned failure in depthmcu_create()
[2019-10-02 11:57:02.088] [error] [t=4447] ../src/sdk/k4a.c (133): depthmcu_create(index, &device->depthmcu) returned failure in k4a_device_open()
libusb: error [_get_usbfs_fd] libusb couldn't open USB device /dev/bus/usb/008/004: Permission denied
libusb: error [_get_usbfs_fd] libusb requires write access to USB device nodes.
[2019-10-02 11:57:02.345] [error] [t=4447] ../src/usbcommand/usbcommand.c (305): find_libusb_device(). Unable to open LIBUSB at index 0 
[2019-10-02 11:57:02.345] [error] [t=4447] ../src/depth_mcu/depth_mcu.c (68): usb_cmd_create(USB_DEVICE_DEPTH_PROCESSOR, device_index, NULL, &depthmcu->usb_cmd) returned failure in depthmcu_create()
[2019-10-02 11:57:02.345] [error] [t=4447] ../src/sdk/k4a.c (133): depthmcu_create(index, &device->depthmcu) returned failure in k4a_device_open()
libusb: error [_get_usbfs_fd] libusb couldn't open USB device /dev/bus/usb/004/004: Permission denied
libusb: error [_get_usbfs_fd] libusb requires write access to USB device nodes.
[2019-10-02 11:57:02.358] [error] [t=4447] ../src/usbcommand/usbcommand.c (305): find_libusb_device(). Unable to open LIBUSB at index 1 
[2019-10-02 11:57:02.358] [error] [t=4447] ../src/depth_mcu/depth_mcu.c (68): usb_cmd_create(USB_DEVICE_DEPTH_PROCESSOR, device_index, NULL, &depthmcu->usb_cmd) returned failure in depthmcu_create()
[2019-10-02 11:57:02.358] [error] [t=4447] ../src/sdk/k4a.c (133): depthmcu_create(index, &device->depthmcu) returned failure in k4a_device_open()
libusb: error [_get_usbfs_fd] libusb couldn't open USB device /dev/bus/usb/008/004: Permission denied
libusb: error [_get_usbfs_fd] libusb requires write access to USB device nodes.
[2019-10-02 11:57:02.559] [error] [t=4447] ../src/usbcommand/usbcommand.c (305): find_libusb_device(). Unable to open LIBUSB at index 0 
[2019-10-02 11:57:02.559] [error] [t=4447] ../src/depth_mcu/depth_mcu.c (68): usb_cmd_create(USB_DEVICE_DEPTH_PROCESSOR, device_index, NULL, &depthmcu->usb_cmd) returned failure in depthmcu_create()
[2019-10-02 11:57:02.559] [error] [t=4447] ../src/sdk/k4a.c (133): depthmcu_create(index, &device->depthmcu) returned failure in k4a_device_open()
libusb: error [_get_usbfs_fd] libusb couldn't open USB device /dev/bus/usb/004/004: Permission denied
libusb: error [_get_usbfs_fd] libusb requires write access to USB device nodes.
[2019-10-02 11:57:02.573] [error] [t=4447] ../src/usbcommand/usbcommand.c (305): find_libusb_device(). Unable to open LIBUSB at index 1 
[2019-10-02 11:57:02.573] [error] [t=4447] ../src/depth_mcu/depth_mcu.c (68): usb_cmd_create(USB_DEVICE_DEPTH_PROCESSOR, device_index, NULL, &depthmcu->usb_cmd) returned failure in depthmcu_create()
[2019-10-02 11:57:02.573] [error] [t=4447] ../src/sdk/k4a.c (133): depthmcu_create(index, &device->depthmcu) returned failure in k4a_device_open()
[... 110 lines cut ...]
OS: Ubuntu 18.04, with kernel 4.16.18-041618-generic
SDK 1.2.0
 - Firmware version:
 == Azure Kinect DK Firmware Tool == 
[... cut ...]
libusb: error [_get_usbfs_fd] libusb requires write access to USB device nodes.
[2019-10-02 12:11:19.641] [error] [t=8835] ../src/usbcommand/usbcommand.c (305): find_libusb_device(). Unable to open LIBUSB at index 1 
[2019-10-02 12:11:19.641] [error] [t=8835] ../src/color_mcu/color_mcu.c (74): usb_cmd_create(USB_DEVICE_COLOR_IMU_PROCESSOR, device_index, NULL, &colormcu->usb_cmd) returned failure in colormcu_create_by_index()
ERROR: Unable to find a device
Found 0 connected devices:

**Additional context**
We also have librealsense installed on the machine.

RoseFlunder, 2019-10-02:
Hi,

did you copy the rules file as described here?
https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/usage.md#linux-device-setup

soalive-smm, 2019-10-02:
Thank you, I had missed that step. That moved the needle, but we're not there yet.

Following moving the udev rules into place, I was able to successfully update the firmware to 1.6.102075014.

However, upon starting the k4aviewer, we're unable to start the device and see the following errors in the shell:

evercoast@evercoast:~/SrcEvercoast/build_MavericksDependencies/Azure-Kinect-Sensor-SDK/build/bin$ ./k4aviewer 
[2019-10-02 12:38:14.759] [error] [t=6514] ../src/dynlib/dynlib_linux.c (86): dynlib_create(). Failed to load shared object libdepthengine.so.2.0 with error: libdepthengine.so.2.0: cannot open shared object file: No such file or directory
[2019-10-02 12:38:14.760] [error] [t=6514] ../src/deloader/deloader.c (70): load_depth_engine(). Failed to Load Depth Engine Plugin (depthengine). Depth functionality will not work
[2019-10-02 12:38:14.760] [error] [t=6514] ../src/deloader/deloader.c (71): load_depth_engine(). Make sure the depth engine plugin is in your loaders path
[2019-10-02 12:38:14.760] [error] [t=6514] ../src/deloader/deloader.c (110): deloader_depth_engine_create_and_initialize(). Failed to load depth engine plugin
[2019-10-02 12:38:14.760] [error] [t=6514] ../src/dewrapper/dewrapper.c (154): depth_engine_start_helper(). Depth engine create and initialize failed with error code: 108.
[2019-10-02 12:38:14.760] [error] [t=6514] ../src/dewrapper/dewrapper.c (160): deresult == K4A_DEPTH_ENGINE_RESULT_SUCCEEDED returned failure in depth_engine_start_helper()
[2019-10-02 12:38:14.760] [error] [t=6514] ../src/dewrapper/dewrapper.c (194): depth_engine_start_helper(dewrapper, dewrapper->fps, dewrapper->depth_mode, &depth_engine_max_compute_time_ms, &depth_engine_output_buffer_size) returned failure in depth_engine_thread()
[2019-10-02 12:38:14.760] [error] [t=6452] ../src/dewrapper/dewrapper.c (552): dewrapper_start(). Depth Engine thread failed to start
[2019-10-02 12:38:14.760] [error] [t=6452] ../src/depth/depth.c (451): dewrapper_start(depth->dewrapper, config, depth->calibration_memory, depth->calibration_memory_size) returned failure in depth_start()
[2019-10-02 12:38:14.765] [error] [t=6452] ../src/depth_mcu/depth_mcu.c (359): cmd_status == CMD_STATUS_PASS returned failure in depthmcu_depth_stop_streaming()
[2019-10-02 12:38:14.765] [error] [t=6452] ../src/depth_mcu/depth_mcu.c (362): depthmcu_depth_stop_streaming(). ERROR: cmd_status=0x00000063
[2019-10-02 12:38:14.769] [error] [t=6452] ../src/sdk/k4a.c (891): depth_start(device->depth, config) returned failure in k4a_device_start_cameras()

[image]

Thanks very much for your help!

soalive-smm, 2019-10-02:
[image]

evercoast-smm, 2019-10-02:
OK, I've completed the installation. The next steps were to install the PPA, as described here:

https://docs.microsoft.com/en-us/windows-server/administration/linux-package-repository-for-microsoft-software

and then install the "depth engine" as described here:
https://docs.microsoft.com/en-us/azure/Kinect-dk/sensor-sdk-download#linux-installation-instructions

Thanks! This can be closed.

soalive-smm, 2019-10-02:
OK, I've completed the installation. The next steps were to install the PPA, as described here:

https://docs.microsoft.com/en-us/windows-server/administration/linux-package-repository-for-microsoft-software

and then install the "depth engine" as described here:
https://docs.microsoft.com/en-us/azure/Kinect-dk/sensor-sdk-download#linux-installation-instructions

Thanks! This can be closed.

closed 2019-10-02 by soalive-smm