microsoft/Azure-Kinect-Sensor-SDK #812: What is Azure Kinect's IR baseline?

msr-peng, 2019-10-03:
Just as the title.

In some robotics application, it's necessary to know the ****IR baseline**(distance between IR projector and IR camera)**, because the IR projector and IR camera form a stereo pair. However, the [Azure Kinect hardware specification page](https://docs.microsoft.com/en-us/azure/kinect-dk/hardware-specification) doesn't has such information. It only has like RGB camera to IR camera.

So what is Azure Kinect's IR baseline distance?

wes-b, 2019-10-04:
You mind sharing how you plan to use the info?

@rabbitdaxi reports that:
> We already do an elliptical correction taking the offset into account when we compute depth map.

The offset you are asking for is:
- NFOV: [-14, 0, 1]
- WFOV [-22, 0, 0]
units are millimeters.

rabbitdaxi, 2019-10-04:
@msr-peng the depth camera is a time-of-flight, therefore, the depth sensor has taken into account of the offset between the laser and sensor on this camera during depth computation. 

I can see structure light camera uses the baseline between IR projector (with dot pattern) and the IR camera for stereo depth estimation. But phase based time-of-flight is a uniform illumination with a completely different principle to compute depth compare to stereo matching.

If you want to be sure, feel free to let us know how you will use above offset information.

closed 2019-12-10 by tesych

martinakos, 2020-01-27:
I have the same question. 

I want to run orbslam2 with the Azure Kinect camera, and I need the baseline for the configuration of orbslam2. 

Orbslam2 in rbgd mode extracts features from the rgb frame and creates a virtual match for a right image, this step needs a baseline. From there, the algorithm continues the same as for a stereo pair.