microsoft/Azure-Kinect-Sensor-SDK #815: Playback of recorded file does not work as expected

wirthual, 2019-10-04:
**Describe the bug**
When trying to replay an recorded file I get the same frame over and over (same timestamp). When I try to access the color data of the recording my app crashes when trying to create a opencv Mat object from the received buffer.

**To Reproduce**
1. Go to https://www.microsoft.com/en-us/download/details.aspx?id=58385&WT.mc_id=
2. Download the OFFICE_SAMPLE_NFOV_UNBINNED.zip
3. Change path to the downloaded file in the following code and execute it:

```C++
#include "iostream"
#include <stdio.h>
#include <k4a/k4a.hpp>
#include <k4arecord/types.h>
#include <k4arecord/playback.hpp>

#include <opencv2/highgui/highgui.hpp>
#include <opencv2/core/core.hpp>
int main(int argc, char **argv) {
    k4a_playback_t playback_handle = NULL;
    if (k4a_playback_open("/home/raphael/Downloads/OFFICE_SAMPLE_NFOV_UNBINNED.mkv", &playback_handle) != K4A_RESULT_SUCCEEDED) {
        printf("Failed to open recording\n");
        return 1;
    }
    k4a::playback record(playback_handle);
    std::chrono::seconds s = std::chrono::duration_cast<std::chrono::seconds>(record.get_recording_length());
    printf("duration of record: %li", s.count());

    k4a::capture cap;
    bool run = true;
    while (run) {
        k4a_wait_result_t result = (k4a_wait_result_t) record.get_next_capture(&cap);
        if (result == K4A_STREAM_RESULT_SUCCEEDED) {
            // Process capture here
            std::cout << "SUCCESS\n";
            k4a::image img = cap.get_depth_image();

            int rows = img.get_height_pixels();
            int cols = img.get_width_pixels();
            std::cout << "depth image size: " << img.get_size() << "\n";
            cv::Mat depthMat(rows, cols, CV_16UC1, (void *) img.get_buffer(), cv::Mat::AUTO_STEP);
            cv::imshow("Bla", depthMat);
            cv::waitKey(1);

            k4a::image img2 = cap.get_color_image();
                int rows2 = img2.get_height_pixels();
                int cols2 = img2.get_width_pixels();
                std::cout << "color image size: " << img2.get_size() << "\n";
                std::cout << "Timestamp: " << img2.get_device_timestamp().count() << "\n";
//        cv::Mat image(cv::Size(rows2,cols2), CV_8UC4);
//        memcpy(image.data, img.get_buffer(),cols2 * rows2 * 4* sizeof(unsigned char));
//        cv::imshow("Bla",image);
//        cv::waitKey(0);
        }
        if (result == K4A_STREAM_RESULT_FAILED) {
            // Process capture here
            std::cout << "FAILED";
        }
        if (result == K4A_STREAM_RESULT_EOF) {
            // Process capture here
            std::cout << "EOF";
            run = false;
        }
    }

}
```
4. Output: 
SUCCESS
depth image size: 737280
color image size: 360619
Timestamp: 10116077
SUCCESS
depth image size: 737280
color image size: 360619
Timestamp: 10116077
SUCCESS
depth image size: 737280
color image size: 360619
Timestamp: 10116077

And so on, timestamp stays the same even though I call get next capture.
I also tried to create the capture inside the while loop, but this leads to a crash of the app 
[2019-10-04 12:10:08.837] [error] [t=20934] ../src/allocator/allocator.c (118): k4a_capture_t_get_context(). Invalid k4a_capture_t (nil)
[2019-10-04 12:10:08.837] [error] [t=20934] ../src/allocator/allocator.c (359): Invalid argument to capture_get_depth_image(). capture_handle ((nil)) is not a valid handle of type k4a_capture_t
[2019-10-04 12:10:08.837] [error] [t=20934] ../src/image/image.c (51): k4a_image_t_get_context(). Invalid k4a_image_t (nil)
[2019-10-04 12:10:08.837] [error] [t=20934] ../src/image/image.c (397): Invalid argument to image_get_height_pixels(). image_handle ((nil)) is not a valid handle of type k4a_image_t
[2019-10-04 12:10:08.837] [error] [t=20934] ../src/image/image.c (51): k4a_image_t_get_context(). Invalid k4a_image_t (nil)
[2019-10-04 12:10:08.837] [error] [t=20934] ../src/image/image.c (390): Invalid argument to image_get_width_pixels(). image_handle ((nil)) is not a valid handle of type k4a_image_t
depth image size: [2019-10-04 12:10:08.837] [error] [t=20934] ../src/image/image.c (51): k4a_image_t_get_context(). Invalid k4a_image_t (nil)
[2019-10-04 12:10:08.837] [error] [t=20934] ../src/image/image.c (366): Invalid argument to image_get_size(). image_handle ((nil)) is not a valid handle of type k4a_image_t
0
[2019-10-04 12:10:08.837] [error] [t=20934] ../src/image/image.c (51): k4a_image_t_get_context(). Invalid k4a_image_t (nil)
[2019-10-04 12:10:08.837] [error] [t=20934] ../src/image/image.c (359): Invalid argument to image_get_buffer(). image_handle ((nil)) is not a valid handle of type k4a_image_t

When I try to execute this for opening a recording:
```
    k4a::playback record;
    record.open("/home/raphael/Downloads/OFFICE_SAMPLE_NFOV_UNBINNED.mkv");
````
then I get:
[2019-10-04 12:15:19.620] [error] [t=22100] ../src/../include/k4ainternal/matroska_read.h (143): k4a_playback_t_get_context(). Invalid k4a_playback_t (nil)
[2019-10-04 12:15:19.620] [error] [t=22100] ../src/record/sdk/playback.cpp (640): Invalid argument to k4a_playback_get_recording_length_usec(). playback_handle ((nil)) is not a valid handle of type k4a_playback_t
duration of record: 0[2019-10-04 12:15:19.620] [error] [t=22100] ../src/../include/k4ainternal/matroska_read.h (143): k4a_playback_t_get_context(). Invalid k4a_playback_t (nil)
[2019-10-04 12:15:19.620] [error] [t=22100] ../src/record/sdk/playback.cpp (449): Invalid argument to k4a_playback_get_next_capture(). playback_handle ((nil)) is not a valid handle of type k4a_playback_t

**Expected behavior**
Get different images when calling get next capture.  

**Desktop:**
 - OS with Version: Ubuntu 18.04.2 LTS
 - SDK Version:   v1.3.0-beta.0-rc (Needed to update to this since I had a problem with the spdlogger in 1.2)
 - Firmware version:
 == Azure Kinect DK Firmware Tool == 
Device Serial Number: 000003492512
Current Firmware Versions:
  RGB camera firmware:      1.6.98
  Depth camera firmware:    1.6.70
  Depth config file:        6109.7
  Audio firmware:           1.6.14
  Build Config:             Production
  Certificate Type:         Microsoft

RoseFlunder, 2019-10-04:
You use the open method the wrong way.
It is a static method that you must not call with an object.
Use:
`k4a::playback handle = k4a::playback::open("/home/raphael/Downloads/OFFICE_SAMPLE_NFOV_UNBINNED.mkv");`

wes-b, 2019-10-04:
Also the readme file has `Synchronized images only: No` which means get_capture can return only a color or a depth image. So you nee d to NULL check to ensure they were captured. If it were yes, then you would not need to NULL check as every capture would have a synchronized depth and color image.

wirthual, 2019-10-04:
Thanks to your help I changed my code to this and got it working:
```C++
#include "iostream"
#include <stdio.h>
#include <k4a/k4a.hpp>
#include <k4arecord/playback.hpp>

#include <opencv2/highgui/highgui.hpp>
#include <opencv2/core/core.hpp>

int main(int argc, char **argv) {
    
    k4a::playback handle =  k4a::playback::open("/home/raphael/Downloads/OFFICE_SAMPLE_NFOV_UNBINNED.mkv");

    std::chrono::seconds s = std::chrono::duration_cast<std::chrono::seconds>(handle.get_recording_length());
    printf("duration of record: %li\n", s.count());

    k4a::capture cap;
    while(handle.get_next_capture(&cap))
    {
        k4a::image img = cap.get_depth_image();

        int rows = img.get_height_pixels();
        int cols = img.get_width_pixels();
        std::cout << "depth image size: " << img.get_size() << "\n";
        std::cout << "Timestamp: " << img.get_device_timestamp().count() << "\n";
        cv::Mat depthMat(rows, cols, CV_16UC1, (void *) img.get_buffer(), cv::Mat::AUTO_STEP);
        cv::imshow("Bla", depthMat);

        k4a::image img2 = cap.get_color_image();
        if(img2){
            int rows2 = img2.get_height_pixels();
            int cols2 = img2.get_width_pixels();
            std::cout << "color image size: " << img2.get_size() << "\n";
            std::cout << "Timestamp: " << img2.get_device_timestamp().count() << "\n";
            std::cout << "width: " << cols2 << "\n";
            std::cout << "heigth: " << rows2 << "\n";
            cv::Mat rawData( 1, img2.get_size(), CV_8UC1, (void*)img2.get_buffer() );
            cv::Mat decodedImage  =  cv::imdecode( rawData,  cv::IMREAD_UNCHANGED );
            cv::imshow("bla",decodedImage);
            cv::waitKey(1);
        }
    }
}
```
Is there an option to use the provided recorder tool to record synchronized files?

And is there an example on how to use the files for playback of a recorded scene?

Thank you very much for your support.

Brent-A, 2019-10-05:
There is no current option to record only synchrnoized frames, but it would be easy to add. It's tracked in issue #557 . It's not exactly what you want, but we do have one [playback example](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/examples/playback_external_sync/main.c).

closed 2019-10-07 by tesych