microsoft/Azure-Kinect-Sensor-SDK #818: Cannot restart multiple synced cameras if RGB stream is OFF in subordinate

DavidRobb, 2019-10-08:
We have an app that requires depth only data from two synced cameras.

As reported in #401, this does not work unless the master device has an RGB stream enabled.

However, it appears that I cannot start, stop and restart streams unless RGB is also enabled on the slave device. In depth only mode, camera times out waiting for frame on restart. Device has to be reset by unplugging and re-plugging. This is the code I use to start each camera. If the line  m_config.color_resolution = K4A_COLOR_RESOLUTION_OFF; is uncommented I cannot stop and restart the cameras. As suggested in the docs, I am starting (and stopping) the slave devices first.

```C++
bool DeviceKinect::start(bool enable_depth) {
    _log->info("start device {} ", m_camId);

    // Manual exposure time recommended for synced depth mode
   int32_t exposureValueUs = 1000;
   int32_t fiftyHz = 1;
   k4a_device_set_color_control(m_device, K4A_COLOR_CONTROL_EXPOSURE_TIME_ABSOLUTE, K4A_COLOR_CONTROL_MODE_MANUAL, exposureValueUs);
   k4a_device_set_color_control(m_device, K4A_COLOR_CONTROL_POWERLINE_FREQUENCY, K4A_COLOR_CONTROL_MODE_MANUAL, fiftyHz);

    m_config.depth_mode = K4A_DEPTH_MODE_WFOV_2X2BINNED;
    m_config.camera_fps = K4A_FRAMES_PER_SECOND_30;
    m_config.color_format = K4A_IMAGE_FORMAT_COLOR_MJPG;
    m_config.color_resolution = K4A_COLOR_RESOLUTION_720P;

    if (K4A_RESULT_SUCCEEDED == k4a_device_get_sync_jack(m_device, &syncInConnected, &syncOutConnected)) {
        if (syncInConnected && !syncOutConnected) {
            _log->info("Sync cable is slave for cam {} ", m_camId);
            m_config.wired_sync_mode = K4A_WIRED_SYNC_MODE_SUBORDINATE;
//            m_config.color_resolution = K4A_COLOR_RESOLUTION_OFF;
            m_config.subordinate_delay_off_master_usec = 160; // Recommended min of 160uS to avoid multiple cam interference
        } else if (!syncInConnected && syncOutConnected) {
            _log->info("Sync cable is master for cam {} ", m_camId);
            m_config.wired_sync_mode = K4A_WIRED_SYNC_MODE_MASTER;
        } else {
            _log->error("Sync cables not correctly connected for cam {} ", m_camId);
            m_config.wired_sync_mode = K4A_WIRED_SYNC_MODE_STANDALONE;
        }
    } else {
        _log->error("Failed to determine Sync cable state for cam {} ", m_camId);
        m_config.wired_sync_mode = K4A_WIRED_SYNC_MODE_STANDALONE;
    }

 
I'm using Windows 10 1903, 

SDK 1.2.0

Device Serial Number: 000541492312
Current Firmware Versions:
  RGB camera firmware:      1.6.102
  Depth camera firmware:    1.6.75
  Depth config file:        6109.7
  Audio firmware:           1.6.14
  Build Config:             Production
  Certificate Type:         Microsoft

Device Serial Number: 000658292412
Current Firmware Versions:
  RGB camera firmware:      1.6.102
  Depth camera firmware:    1.6.75
  Depth config file:        6109.7
  Audio firmware:           1.6.14
  Build Config:             Production
  Certificate Type:         Microsoft

wes-b, 2019-10-08:
How many devices do you have linked together? I am wondering if this is an issue with a subordinate device driving another subordinate. Where the clock pulse is to the subordinates child is missing because the color camera is not enabled.

DavidRobb, 2019-10-08:
Just two devices. One master and one subordinate with a single cable linking them

wes-b, 2019-10-08:
Confirmed, I can repro this too.

wes-b, 2019-10-09:
This feature worked in firmware AzureKinectDK_Fw_1.6.987014 and regressed in AzureKinectDK_Fw_1.6.102075014.bin. There are lost of multi camera fixes after 1.6.987014 so I don't recommend going back.

PR #1115 (2020-03-09): Enabling depth only multi device test

## Fixes #818

### Description of the changes:
- This is fixed in Firmware 1.6.108079014.bin
closed 2020-03-09 by wes-b