microsoft/Azure-Kinect-Sensor-SDK #828: transformation_depth_image_to_color_camera_validate_parameters has switched expected / actual parameters

nvibd, 2019-10-10:
In `rgbz.c`, the function `transformation_depth_image_to_color_camera_validate_parameters` makes use of `transformation_compare_image_descriptors`. However, `transformation_compare_image_descriptors` has the parameters switched between the expected and actual values compared to the usage in `transformation_depth_image_to_color_camera_validate_parameters`.

E.g., the code currently is like this:
```
if (transformation_compare_image_descriptors(depth_image_descriptor, &expected_depth_image_descriptor) == false)
    {
        LOG_ERROR("Unexpected depth image descriptor, see details above.", 0);
        return K4A_BUFFER_RESULT_FAILED;
    }
```

But expected / actual should be switched:
```
if (transformation_compare_image_descriptors(&expected_depth_image_descriptor, depth_image_descriptor) == false)
    {
        LOG_ERROR("Unexpected depth image descriptor, see details above.", 0);
        return K4A_BUFFER_RESULT_FAILED;
    }
```

There are several usages of `transformation_compare_image_descriptors` where this is the case.

I noticed this when debugging and the debug output greatly confused me as I was looking why the expected value is wrong (instead of the actual one).

wes-b, 2019-10-10:
Thanks for filing the issue. `transformation_compare_image_descriptors` doesn't assuming anything about the descriptors passed in. In fact both parameters are const, so I don't see what the issue is.

nvibd, 2019-10-11:
The debug output reads as "expected values: ... / actual values: ...". However, the actual values (with which the function is called) are displayed under "expected values" and vice versa, leading to confusion when debugging.

PR #834 (2019-10-11): Fixing wrong error message
## Fixes #828

### Description of the changes:
- Fixed broken error message
closed 2019-10-11 by wes-b