microsoft/Azure-Kinect-Sensor-SDK #829: Body tracking depth limit?

RenderHeadsMrT, 2019-10-10:
Hi,

I'm not actually sure of the maximum number of tracked bodies (is this known?) but in certain situations there may be a crowd of people watching a single person interact and I can see situations where it tracks X people in the crowd before failing to track the intended user.

There are obvious physical restrictions that could be put in place (camera angle, crowd control etc..) but it would be good if we could tell the tracker to ignore anything past a certain depth (X meters) to further control situations such as this.

I'm not missing any features that allow this currently? And is there any form of hard depth limit as is? (I'm using 0.9.2 as 0.9.3 seems to be less stable for our tracking usage) If not, is this something that could be considered? (And maybe further down the road cropped regions of interest, masking etc..)

Cheers,
Richard

UnaNancyOwen, 2019-10-11:
I have done try it. It may be helpful for you.
The body tracking process are detects 2d human pose from infrared image in the first step.
Therefor, the body tracking area can be limited by setting Infrared to 0 (It is not detected area) according to the depth value, image area, and other. (NOTE: It is necessary to edits the buffer directly. And, it needs to be done before enqueue.)
https://twitter.com/UnaNancyOwen/status/1167443753365192704

RenderHeadsMrT, 2019-10-11:
Hi Tsukasa,

That's exactly what I was looking for, thanks! Didn't spot that we could manipulate the actual buffer data fed into the tracker (thought it was a copy). So far seems to work great.

Cheers,
Richard

closed 2019-10-14 by wes-b

brunzero, 2019-12-13:
Is there any further insight here as to the total amount of tracked bodies per device?