microsoft/Azure-Kinect-Sensor-SDK #830: k4arecorder only records the color camera

monajalal, 2019-10-10:
why the following only captures the color camera?

```
C:\Program Files\Azure Kinect SDK v1.2.0\tools>k4arecorder.exe -d WFOV_UNBINNED -c 3072p -r 15 -l 10 --imu OFF  C:\Users\mona6\workspace\datacapture\output.mkv
Device serial number: 000885793812
Device version: Rel; C: 1.6.102; D: 1.6.75[6109.7]; A: 1.6.14
Device started
Started recording
Stopping recording...
Saving recording...
Done
```

RoseFlunder, 2019-10-10:
In which way did you confirm that there is no depth track?
Did you try to open the mkv with the k4aviewer?

monajalal, 2019-10-10:
fixed. we used viewer and fixed.
[image]

closed 2019-10-10 by monajalal