microsoft/Azure-Kinect-Sensor-SDK #836: sudo apt-get install libk4a1.1-dev and  sudo apt-get install k4a-tools fail

muzizhi, 2019-10-12:
when I try to run:sudo apt-get install libk4a1.1-dev
some error happened:
Unpacking libk4a1.3 (1.3.0) ...
dpkg: error processing archive /var/cache/apt/archives/libk4a1.3_1.3.0_amd64.deb (--unpack):
 trying to overwrite '/usr/lib/x86_64-linux-gnu/libk4a.so', which is also in package libk4a1.1-dev 1.1.1
Errors were encountered while processing:
 /var/cache/apt/archives/libk4a1.3_1.3.0_amd64.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)

so any advice?

RoseFlunder, 2019-10-12:
It seems you have already installed version 1.1 when I interpret the message correctly and afterwards it fails to install 1.3 because 1.1 is already installed.

It is not possible to have multiple different versions of the sdk installed.
So when you try to install a specific version you have to uninstall all other versions first.
The sdk itself has its version in the package name but the k4a-tools package has a version number.
You can install a specific version of the tools package with "sudo apt install k4a-tools=1.2" when you want the tools for version 1.2 for example.
Otherwise it will try to install the latest version with needs sdk version 1.3.

So in your case it tries to install sdk 1.3 for the tools which is not possible because you already have 1.1, so the installation fails.

muzizhi, 2019-10-14:
understand，thanks

schultetwin1, 2019-10-14:
@RoseFlunder, thank your for the help! You are correct, but unfortunately that was not our designed. For more info see below.

We have three different packages

* `libk4a` (our runtime package)
* `libk4a-dev` (out development package)
* `k4a-tools` (our tools package)

Each version of `k4a-tools` and `libk4a-dev` take a dependency on a version of `libk4a`.

For example:

`k4a-tools` version 1.3.0 depends on `libk4a1.3`. Similarly, `libk4a1.3-dev` version 1.3.1 (when it comes out) will depend on version `libk4a1.3`.

We have designed each different version of `libk4ax.y` to be installed next to each other. For example, you should be able to install `libk4a1.2` and `libk4a1.3` next to each other. (Note, this is currently broken. Keep reading).

This design is not true for our dev packages and our tools packages. You should **not** be able to install `k4a-tools` version 1.2.0 and `k4a-tools` version 1.3.0 next to each other. If you want to install `k4a-tools` version 1.3.0 you need to remove `k4a-tools` 1.2.0. `apt` should do that for you. The same is true for `libk4a1.3-dev`. If you want to install `libk4a1.3-dev` you need to uninstall `libk4a1,2-dev`. `apt` will **not** do that for you.

Currently there is a bug in this design for `libk4a`. If you try to install `libk4a1.2` and `libk4a1.3` you'll get the following error:

>Unpacking libk4a1.3 (1.3.0) ...
>dpkg: error processing archive /var/cache/apt/archives/libk4a1.3_1.3.0_amd64.deb (--unpack):
> trying to overwrite '/usr/lib/x86_64-linux-gnu/libdepthengine.so.2.0', which is also in package libk4a1.2 1.2.0
>dpkg-deb: error: paste subprocess was killed by signal (Broken pipe)
>Errors were encountered while processing:
> /var/cache/apt/archives/libk4a1.3_1.3.0_amd64.deb
>E: Sub-process /usr/bin/dpkg returned an error code (1)

This is because the depthengine in both `libk4a1.2` and `libk4a1.3` conflict. This needs to be fixed. I will use this bug to track this.

iiwolf, 2019-11-07:
> You can install a specific version of the tools package with "sudo apt install k4a-tools=1.2" when you want the tools for version 1.2 for example.

Should this work? I receive the following error:
`E: Version '1.2' for 'k4a-tools' was not found`

So at this point, is there anyway to have the body tracking sdk and viewer installed alongside each other?

Thanks

iiwolf, 2019-11-07:
Never mind! You just have to specify 1.2.0

`sudo apt install k4a-tools=1.2`

hrsFire, 2020-04-03:
Interestingly, it is the same problem as #1153 . But my issue was closed without further comment. When will this issue be solved?

Fugashu, 2020-04-15:
Try to uninstall the version you are currently trying to install. Also make sure to uninstall all older versions of k4a-tools. Then try a clean install again, in my case that worked.

themancalledjakob, 2020-05-21:
you can also install the sdk in a local directory per project, without touching a system wide install.
 ```
mkdir localk4a
wget https://packages.microsoft.com/ubuntu/18.04/prod/pool/main/libk/libk4a1.4/libk4a1.4_1.4.0_amd64.deb
wget https://packages.microsoft.com/ubuntu/18.04/prod/pool/main/libk/libk4a1.4-dev/libk4a1.4-dev_1.4.0_amd64.deb 
dpkg -x ./libk4a1.4_1.4.0_amd64.deb ./localk4a/
dpkg -x ./libk4a1.4-dev_1.4.0_amd64.deb ./localk4a/
```

You need to then take care of including and linking the SDK properly, but it allows to have a system wide install of k4a-tools and libk4a which is up to date, and a less recent one for body tracking for specific projects.
So far this works for me.

Since Linux support for libk4abt is lagging behind, and makes a setup like this often necessary, it would be great if this would be fixed in the debian package repository though.

Are there more issues than conflicting symbolic links like `/usr/lib/x86_64-linux-gnu/libk4a.so -> /usr/lib/x86_64-linux-gnu/libk4a.so.1.4` & `/usr/lib/x86_64-linux-gnu/libk4a.so -> /usr/lib/x86_64-linux-gnu/libk4a.so.1.3`?

sewanfan, 2020-05-31:
Hi Jacob and others,

Thanks for tip on installing to a local folder.  Tested to work ok. Also installed system wide.

Installed MS 99 udev rules also in /etc/udev/rules

Our machine is linux mint virtual machine generic #34~18.04.2-Ubuntu running on Windows 10 Pro

Been reading the git hub forum pages to follow the 3 pieces of the installation and completed them

We use pmd pico flexx camera with little problem on sdk installation and it seems that MS kinect4a sdk installation is all over the place.

We don't have a kniect4a camera yet but would like learn all the installation bugs before buying one from Amazon ($599).  Our institution not buy from MS.

Can any body in this forum provide simple compile line or a link reference for simple gcc to compile and test simple program using sdk libraries?

something like 

gcc kinecttest1.c -llib...

Already tested the #include <k4a/k4a.h> to work in simple c program ok.

Thank you for your time

sewanfan, 2020-06-01:
Found the link below together with Jacob's comment for making a local directory to expose where the .h and .so files are installed

https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/652#issuecomment-608118432

The three MS kinect sdk files need to install, remove and install again to avoid the version errors mentioned in the first comments

We use pmd pico flexx time of flight camera and their software is all in one place not all over the place like that in kinect4a.  

But kinect4a is for developers not for regular users so that is probably why all over the place.

We use pmd camera to develop computer vision for a robotic strawberry harvester.

We like to use kinect4a camera with its dual camera without having to add another camera to pmd pico flexx TOF camera.

With our pmd pico flexx plugged into usb and when you do dmesg there is no linux device file like /dev/video0 for example.  

But their viewer stills works and shows the gray scale and distance images just like for the kinect4a examples and can use OpenCV libraries no problem

Similarly, in another post for kinect4a, from their dmesg there is also no linux device file attached.

Buy they must be using video for linux 2 somewhere may be

Does any one know some details as how the kinect4a camera gets its linux device file if it uses one at all?

Below simple c program to test out the kinect4a sdk and has the gcc compile in the comment 

/* 5/30/20
testing header file for Microsoft kinect 4a if works or not
5/31/20
kinect header files are in /usr/include/k4a
installed
compiles ok, no errors
to compile, use
gcc test_kinect4a1.c
works to compile ok
to compile with k4a library use
gcc test_kinect4a1.c -lk4a
*/

#include <stdio.h>
#include <stdlib.h>
#include <k4a/k4a.h>
/*#include <k4arecord/record.h>*/

int main()
{
  printf("ok\n");

  uint32_t count = k4a_device_get_installed_count();
    if (count == 0)
    {
        printf("No k4a devices attached!\n");
        return 1;
    }
    return(0);
} /* end main */

output below as expected since we don't yet own a kniect4a camera

sfan@sfan-VirtualBox:~/c/work20201/kinect4a1$ ./a.out 
ok
No k4a devices attached!

Thank you for your time

adrianramospeon, 2021-01-14:
More than a year has passed since the issue was opened. Any plans on fixing this? Development seems stalled :(

Update on @iiwolf's  solution to use k4a-tools and libk4abt1.0 simultaneously now that libk4a1.4 is out:

`sudo apt install libk4abt1.0 k4a-tools=1.3.0`

> Never mind! You just have to specify 1.2.0
> 
> `sudo apt install k4a-tools=1.2.0`