microsoft/Azure-Kinect-Sensor-SDK #838: Ubuntu packages depend on with sensor sdk 1.2.0 instead 1.3.0

RoseFlunder, 2019-10-14:
The recently released body tracking sdk 0.9.4 (package libk4abt0.9 in version 0.9.4) still depends on sensor sdk 1.2.0 (libk4a1.2).
Because the sensor sdk has a new seperate package for each version the dependency must be changed to "libk4a1.3" instead of "libk4a1.2".

[image]

Moreover the "libk4abt-dev" package is still on version 0.9.3.

[image]

So all in all

yijiew, 2019-10-14:
@RoseFlunder Thanks for reporting this issue. I reuploaded the "libk4abt0.9-dev" 0.9.4 version. Please give it a try to see whether it resolves this part of the issue. As for the request of updating the sensor SDK dependency to 1.3.0, we will have to have another release for this. Just out of curiosity, is there any feature that you relies on only existing in Sensor SDK 1.3.0?

RoseFlunder, 2019-10-14:
@yijiew 
I am using the SDK in a ROS environment together with this package:
https://github.com/microsoft/Azure_Kinect_ROS_Driver

When using the ros node from the package together with sdk 1.2.0 in a nodelet (multiple ros nodes in one process) a problem can occur because the sdk exported methods from the static linked libjpegturbo which was reported here:
https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/672

So if other nodes in the same process make use of libjpegturbo but dynamically linked to another version a symbol conflict occurs.
One example would be starting this example launch file:
https://github.com/microsoft/Azure_Kinect_ROS_Driver/blob/melodic/launch/rectify_test.launch

This problem got fixed with 1.3.0 as you can see in the linked merge request:
https://github.com/microsoft/Azure-Kinect-Sensor-SDK/pull/763

yijiew, 2019-10-14:
@RoseFlunder Thanks for your detailed explanation. We will make sure we link to the latest sensor SDK in our next release. Please stay tuned :)

RoseFlunder, 2019-10-15:
Thanks.
I can confirm that the k4abt0.9-dev in version 0.9.4 provides the correct header files and works fine together with the k4abt0.9 package.

SongJuHeon, 2019-10-25:
Hi,

I added the repository for my OS version by checking the link below: 
https://docs.microsoft.com/ko-kr/windows-server/administration/linux-package-repository-for-microsoft-software

And then, I tried to reinstall the"libk4abt0.9-dev" 0.9.4 version after installing libk4a1.3.
However, I can't install libk4abt0.9(0.9.4) (the same issue still appears).
The result of entering the code 'sudo apt show libk4abt0.9 ' is the same as what @RoseFlunder already explained (depenc libk4a1.2).

Is there anyways to resolve this problem? or Something that I missed for setting repository(additional repository) ?? 

Furthermore, after adding the repository mentioned above, message related to i386 issue suddenly appears at updating process.
Is it okay to ignore this message and continue? 

The following is the environment that I'm currently working on:
OS: ubuntu 18.04 LTS
Date: 19-10-25
amd64

RoseFlunder, 2019-10-25:
The only way to use body tracking sdk on ubuntu at the moment is in combination with sensor sdk 1.2.0.

bearpaw, 2019-10-31:
> @yijiew
> I am using the SDK in a ROS environment together with this package:
> https://github.com/microsoft/Azure_Kinect_ROS_Driver
> 
> When using the ros node from the package together with sdk 1.2.0 in a nodelet (multiple ros nodes in one process) a problem can occur because the sdk exported methods from the static linked libjpegturbo which was reported here:
> #672
> 
> So if other nodes in the same process make use of libjpegturbo but dynamically linked to another version a symbol conflict occurs.
> One example would be starting this example launch file:
> https://github.com/microsoft/Azure_Kinect_ROS_Driver/blob/melodic/launch/rectify_test.launch
> 
> This problem got fixed with 1.3.0 as you can see in the linked merge request:
> #763
> 
> EDIT: So I am more interested in the bug fix from 1.3.0 rather than a new feature. But currently I can work without using nodelets and switch to nodelets when there is a new body tracking release then.

Hi @RoseFlunder , could you please share how you run the body tracking without nodelets? Do you write your own launch file? Thank you in advance!

RoseFlunder, 2019-10-31:
Sure, I just use this launch file and set body_tracking_enabled to true and a depth_mode of your choice:
https://github.com/microsoft/Azure_Kinect_ROS_Driver/blob/melodic/launch/driver.launch

qm13, 2019-12-07:
Fixed in v0.9.5.

closed 2019-12-07 by qm13