microsoft/Azure-Kinect-Sensor-SDK #839: Crashed when I start the device

perelmangao, 2019-10-14:
I installed the SDK according to the instructions and the device can be detected. When I start camera, it crashed with an errror. 
`
[ warning  ] : imu_capture_ready(). IMU startup dropped last 408 samples, the timestamp is too large
[ warning  ] : depth_engine_thread(). Depth image processing is too slow at 432ms (this may be transient).
[ warning  ] : capturesync_add_capture(). Dropping depth capture as TS is too large TS:1071084400
[ warning  ] : queue_pop(). dewrapper: Dropped oldest 7 captures from queue
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:     17733 type:Depth
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:     51066 type:Depth
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:     84400 type:Depth
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    117733 type:Depth
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    151066 type:Depth
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    184400 type:Depth
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    217733 type:Depth
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    251066 type:Depth
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    284400 type:Depth
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    317733 type:Depth
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    351066 type:Depth
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    384400 type:Depth
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    417733 type:Depth
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    451066 type:Depth
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    484400 type:Depth
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    517733 type:Depth
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    551066 type:Depth
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    584400 type:Depth
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    617733 type:Depth
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    651066 type:Depth
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    684400 type:Depth
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    717733 type:Depth
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    751066 type:Depth
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    784400 type:Depth
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    817733 type:Depth
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    851066 type:Depth
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    884400 type:Depth
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    917733 type:Depth
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    951066 type:Depth
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:    984400 type:Depth
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ error    ] : replace_sample(). capturesync_drop, releasing capture early due to full queue TS:   1017733 type:Depth
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ warning  ] : CFrameContext(). No metadata attached to the sample
[ warning  ] : queue_disable(). Waiting for blocking call to complete
[ warning  ] : queue_error(). Error detected, shutting down queue and notifying consumers
[ warning  ] : queue_error(). Error detected, shutting down queue and notifying consumers
[ warning  ] : queue_error(). Error detected, shutting down queue and notifying consumers
[ warning  ] : capturesync_add_capture(). Capture Error Detected, Depth 
[ warning  ] : CFrameContext(). No metadata attached to the sample
`
[image]

- OS is windows10
- CPU is intel i5-7200U @2.5GHz
- RAM is 8G
- DirectX12

wes-b, 2019-10-14:
It is not a crash (at least not from the info I see here), but does look like an error starting the color camera's. 

Can you look in the event viewer to see if there are any services like "Frame Server" crashing? In the past issues like this have root caused other system components present on the PC. So review your PC's installed software and see if there is any camera related software that can be removed.

The messages about CFrameContext is unique, take a look at issue #459. We didn't resolve the issue there, but there is some additional trouble shooting steps.

perelmangao, 2019-10-15:
> It is not a crash (at least not from the info I see here), but does look like an error starting the color camera's.
>
> Can you look in the event viewer to see if there are any services like "Frame [...]

 Hi, I've checked the event viewer. It shows that Windows Camera Frame Server is running.

wes-b, 2019-10-15:
Can you capture an "Info" level log for this issue? I would like to see if there are signs it might be opening the wrong camera.

I still suspect another camera or its software may be a factor in this issue. Can you try disabling other cameras in Device Manager? If that still doesn't work can you try to uninstall other camera related software and drivers?

wes-b, 2019-11-06:
We can reopen this if you still have the issue. Anyone else seeing something similar, please open a new issue.

closed 2019-11-06 by wes-b

yeongwoonIm, 2019-12-20:
> We can reopen this if you still have the issue. Anyone else seeing something similar, please open a new issue.

I think I have a similar problems. I open a new issue
https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/998

wes-b, 2019-12-20:
Please share the full log

#998<https://nam06.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fmicrosoft%2FAzure-Kinect-Sensor-SDK%2Fissues%2F998&data=02%7C01%7CWes.Barcalow%40microsoft.com%7C20de959e54c6441b066308d7851a08d8%7C72f988bf86f141af91ab2d7cd011db47%7C1%7C0%7C637124219171151002&sdata=8uXojX6YKHFhvb7P%2BMJtMZyieu22sfkTCvTGxAlVqgI%3D&reserved=0>

yeongwoonIm, 2019-12-20:
> Please share the full log Get Outlook for Android<https://aka.ms/ghei36>
> […](#)

I updated with full log https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/998