microsoft/Azure-Kinect-Sensor-SDK #840: Transforming depth to color results in loss of depth data on certain devices

joshberry, 2019-10-14:
**Describe the bug**
Transforming a capture from the depth camera perspective to the color camera perspective introduces holes that weren't present in the source depth map when executing on certain devices. 

```
// C# example
transform.DepthImageToColorCamera(capture, transformedDepth);
```

**To Reproduce**
Run the [TransformDepthToColor](https://github.com/microsoft/Azure-Kinect-Samples/tree/master/build2019/csharp/2%20-%20TransformDepthToColor) solution in the [Azure-Kinect-Samples](https://github.com/microsoft/Azure-Kinect-Samples) repo on a (or at least mine) Surface Laptop 2. The resulting image is full of holes that did not exist in the raw depth map.

[image]

Running the same program on any other device we've tested results in a transformed image with minimal holes (just like the source depth map).

[image]

**Expected behavior**
All devices with supported hardware should have the same results when transforming depth to color image.

**Logs**

**Screenshots**

**Desktop (please complete the following information):**
 - OS with Version: 1809 17763.805
 - SDK Version:  1.3.0
 - Firmware version:
  RGB camera firmware:      1.6.98
  Depth camera firmware:    1.6.70
  Depth config file:        6109.7
  Audio firmware:           1.6.14
  Build Config:             Production
  Certificate Type:         Microsoft

**Additional context**
So far, we've only been able to produce this issue on a Surface Laptop 2 (Intel Core i5 / 8GB RAM). All other devices we've tested have worked as expected.

I've compared the original depth maps for the first and second examples, and they are virtually identical, which suggests the holes are being created as part of the transformation. Not sure why this would be hardware dependent though.

rabbitdaxi, 2019-10-14:
@joshberry this is likely the graphic card driver issue. Please make sure you install the latest graphic card driver (since you are using surface laptop, this might be related to the intel GPU driver update was not automatically update to latest by the windows update). Please refer to this issue see whether the comment there helps you #294

joshberry, 2019-10-15:
@rabbitdaxi That did the trick. Thanks for your help!

closed 2019-10-15 by joshberry

samoteph, 2019-11-06:
Here the process to force installation of recent intel driver on Surface devices : https://dancharblog.wordpress.com/2018/12/09/installing-intel-uwd-gpu-drivers-on-surface-and-other-oem-laptops/