microsoft/Azure-Kinect-Sensor-SDK #842: Allow the viewer to also save the displayed data

shreyapandit, 2019-10-15:
**Is your feature request related to a problem? Please describe.**

The viewer functionalities do not seem to support actual saving of data. That happens separately through the recorder, which seems redundant because most of the config options can be set through the viewer itself.

**Describe the solution you'd like**

 Since the viewer gets all the frames anyways for display, it should also allow save of data through the Start/Stop Button.

**Describe alternatives you've considered**

Currently recording through Command Line recorder

**Additional context**

shreyapandit, 2019-10-15:
Looks like the code is open sourced: https://github.com/microsoft/Azure-Kinect-Sensor-SDK/tree/95f1d95f1f335b57a350a80a3a62e98e1ee4258d/tools/k4aviewer.

If someone can point me to where the Start/Stop UI functions are, I can trace my way back and try to add this and open a PR :) 

Cheers and once again thanks for all your hard work on this!

shreyapandit, 2019-10-15:
Hi, I am trying to build the SDK from scratch. While the SDK builds without errors, I am unable to use any application that uses the depth engine. For example, in the fastcapture streaming example, here is the error I get:

``` ** fastcapture_streaming.exe -help for usage information
 ** -----------------------------------------------------
.  <Directory>
60  <Streaming Length in seconds>
[2019-10-15 09:28:48.034] [error] [t=10000] ..\..\src\dynlib\dynlib_windows.c (115): dynlib_create(). Failed to load DLL depthengine_2_0 with error code: 126
[2019-10-15 09:28:48.036] [error] [t=10000] ..\..\src\deloader\deloader.cpp (75): deloader_init_once(). Failed to Load Depth Engine Plugin (depthengine). Depth functionality will not work
[2019-10-15 09:28:48.039] [error] [t=10000] ..\..\src\deloader\deloader.cpp (76): deloader_init_once(). Make sure the depth engine plugin is in your loaders path
[2019-10-15 09:28:48.039] [error] [t=10000] ..\..\src\deloader\deloader.cpp (113): deloader_depth_engine_create_and_initialize(). Failed to load depth engine plugin
[2019-10-15 09:28:48.039] [error] [t=10000] ..\..\src\dewrapper\dewrapper.c (154): depth_engine_start_helper(). Depth engine create and initialize failed with error code: 108.
[2019-10-15 09:28:48.040] [error] [t=10000] ..\..\src\dewrapper\dewrapper.c (160): deresult == K4A_DEPTH_ENGINE_RESULT_SUCCEEDED returned failure in depth_engine_start_helper()
[2019-10-15 09:28:48.040] [error] [t=10000] ..\..\src\dewrapper\dewrapper.c (194): depth_engine_start_helper(dewrapper, dewrapper->fps, dewrapper->depth_mode, &depth_engine_max_compute_time_ms, &depth_engine_output_buffer_size) returned failure in depth_engine_thread()
[2019-10-15 09:28:48.041] [error] [t=1012] ..\..\src\dewrapper\dewrapper.c (552): dewrapper_start(). Depth Engine thread failed to start
[2019-10-15 09:28:48.042] [error] [t=1012] ..\..\src\depth\depth.c (451): dewrapper_start(depth->dewrapper, config, depth->calibration_memory, depth->calibration_memory_size) returned failure in depth_start()
[2019-10-15 09:28:48.043] [error] [t=1012] ..\..\src\depth_mcu\depth_mcu.c (359): cmd_status == CMD_STATUS_PASS returned failure in depthmcu_depth_stop_streaming()
[2019-10-15 09:28:48.043] [error] [t=1012] ..\..\src\depth_mcu\depth_mcu.c (362): depthmcu_depth_stop_streaming(). ERROR: cmd_status=0x00000063
[2019-10-15 09:28:48.044] [error] [t=1012] ..\..\src\sdk\k4a.c (891): depth_start(device->depth, config) returned failure in k4a_device_start_cameras()
[Streaming Service] Failed to start cameras
Configuration Failed.
[Streaming Service] Stopped.```

Would love advice as to how I can get past this? I am facing the same issue while working with the k4aviewer from scratch.

shreyapandit, 2019-10-15:
I resolved the above by looking at another issue, where some one had mentioned I can copy the depth engine dll over to where the k4a.dll is and then the error went away

wes-b, 2019-10-15:
@shreyapandit thanks for the update. If you have any feedback on how to make this easier for others please feel free to share. Also would you mind sharing what your workflow was to get into the state of not knowing you needed the depth engine? I am guessing you built the SDK yourself? Were you referring to any documentation in particular?

wes-b, 2019-10-15:
> Is your feature request related to a problem? Please describe.
>
> The viewer functionalities do not seem to support actual saving of data. That happens separately through the recorder, which seems [...]

We would be open to this functionality being added to viewer and encourage the community to help out.

1216704928, 2019-11-19:
Is it possible to make the viewer save video？