microsoft/Azure-Kinect-Sensor-SDK #843: Viewer doesn't launch

yeongwoonIm, 2019-10-16:
I launch the Azure Viewer. but it doesn't work..

OS : windows 10 enterprise 2016 LTSB
CPU : intel i7-6700K cpu @ 4.00GHz
RAM : 8 GB
GPU : HD530

windows event viewer show logs
오류 있는 응용 프로그램 이름: k4aviewer.exe, 버전: 1.2.0.14, 타임스탬프: 0x5d5dcd1b
오류 있는 모듈 이름: ig9icd64.dll, 버전: 25.20.100.6323, 타임스탬프: 0x5ba3fd9c
예외 코드: 0xc0000005
오류 오프셋: 0x000000000001f82e
오류 있는 프로세스 ID: 0x1708
오류 있는 응용 프로그램 시작 시간: 0x01d583e0847d7752
오류 있는 응용 프로그램 경로: C:\Program Files\Azure Kinect SDK v1.2.0\tools\k4aviewer.exe
오류 있는 모듈 경로: C:\Windows\System32\DriverStore\FileRepository\igdlh64.inf_amd64_8a9535cd18c90bc3\ig9icd64.dll
보고서 ID: 42a9b58c-551a-47c1-8e4b-f7bc4e996f06
오류 있는 패키지 전체 이름: 
오류 있는 패키지에 상대적인 응용 프로그램 ID:

wes-b, 2019-10-24:
Translated your log says: 

Windows event viewer show logs
Application name with error: k4aviewer.exe, version: 1.2.0.14, timestamp: 0x5d5dcd1b
Error module name: ig9icd64.dll, version: 25.20.100.6323, timestamp: 0x5ba3fd9c
Exception code: 0xc00000005
Error offset: 0x0000000001f82e
Process ID with error: 0x1708
Application start time with error: 0x01d583e0847d7752
Application path with error: C:Program FilesAzure Kinect SDK v1.2.0toolsk4aviewer.exe
Error module path: C:WindowsSystem32DriverStoreFileRepositoryigdlh64.inf_amd64_8a9535cd18c90bc3ig9icd64.dll
Report ID: 42a9b58c-551a-47c1-8e4b-f7bc4e996f06
The full name of the package with the error:
Application IDs relative to the package that is error:

wes-b, 2019-10-24:
Have you tried uninstalling the software/driver that installed ig9icd64.dl? 

If this is related to remote desktop this could be #503.

yeongwoonIm, 2019-10-25:
I tried update Intel graphic driver. then it shows latest version of driver.
but I didn't try uninstall driver...;;; cuz the PC far away from my office.
and I tried other PC that was remote connected.
PC has a graphic card GTX 1050.
I don't think it is related with remote desktop problems...
so I ordered 7th Intel CPU. and it will test next week.

wes-b, 2019-11-06:
@yeongwoonlm any update?

closed 2019-11-08 by wes-b

yeongwoonIm, 2019-12-12:
> @yeongwoonlm any update?

I upgrade CPU 6th Intel CPU to 7th Intel CPU yesterday.
at first time, it worked with a little problem that RGB Camera didn't work.
but... today, it doesn't work Viewer too.
error Message is same things. => ig9icd64.dll
==> error Messages.
오류 있는 응용 프로그램 이름: k4aviewer.exe, 버전: 1.3.0.23, 타임스탬프: 0x5d98be78
오류 있는 모듈 이름: ig9icd64.dll, 버전: 25.20.100.6323, 타임스탬프: 0x5ba3fd9c
예외 코드: 0xc0000005
오류 오프셋: 0x000000000001f82e
오류 있는 프로세스 ID: 0x1220
오류 있는 응용 프로그램 시작 시간: 0x01d5b0bd539dc00f
오류 있는 응용 프로그램 경로: C:\Program Files\Azure Kinect SDK v1.3.0\tools\k4aviewer.exe
오류 있는 모듈 경로: C:\Windows\System32\DriverStore\FileRepository\igdlh64.inf_amd64_8a9535cd18c90bc3\ig9icd64.dll
보고서 ID: f19c1f3a-5147-4cd8-93da-f46591a7cae5
오류 있는 패키지 전체 이름: 
오류 있는 패키지에 상대적인 응용 프로그램 ID: 

I made my app. it works very well.
The app use only Depth Datas.
[image]

I think Viewer Has a bugs..
I hope it will repaire soon..

wes-b, 2019-12-12:
@yeongwoonlm the error is that ig9icd64.dll is crashing. This is not an Azure Kinect binary. It is part of the software already present on your PC. You need to investigate how to remove it or update it.

yeongwoonIm, 2019-12-13:
I figure out launching Viewer
step 1.
right click k4aviewer.
select properties.
select compatibility and Run the compatibility troubleshooter
[image]

step 2.
Run Recommended setting.
[image]

step 3.
Run program test.
[image]

may be you can see launched Viewer
[image]

click Yes. 
[image]

Done~

I hope this comment helps

Thank you for Considering my issue. ^_^ @wes-b

wes-b, 2019-12-17:
Sorry, we don't have any way to translate this. I was able to translate the earlier error because I could copy your text. I can not copy the text from the screen images though.

wes-b, 2019-12-17:
It looks like you used compatibility mode to solve your crash, good to know this is an option.

yeongwoonIm, 2019-12-19:
> Sorry, we don't have any way to translate this. I was able to translate the earlier error because I could copy your text. I can not copy the text from the screen images though.

There is one way translate images on other languages,  using App of Google Translator on smartPhone. 

[https://play.google.com/store/apps/details?id=com.google.android.apps.translate&hl=ko](url)

it working like this
[image]

anyway, you are right. I used compatibility mode.