microsoft/Azure-Kinect-Sensor-SDK #846: Low volume of the microphone array on Ubuntu

mgeletka, 2019-10-18:
Hi, 
I want to access the Kinect Microphone array from the Ubuntu 18.04 (without the SDK), the microphone is recognized but volume  is significantly lower compared to the microphone connected to Windows 10. When I compared two spectograph from Ubuntu and Windows they differed in 20 dB. 

I tried connect Kinect to different PC with Ubuntu installed but always ended up with the same result. On other hand if I tried it on the Windows it worked without any problems.

In the PulseAudio and the Alsamixer is the microphone on the maximum, but as show in the screenshot below the intensity of the Kinect Microphone array to my internal Microphone is very low.

**Output from PulseAudio**
[image]

**Ouput from alsamixer**
[image]the
[image]
[image]

I tried to record audio from Python (PyAudio) and from C libsoundio. In both cases the volume of the output was the same. 
**The list of input devices from ./sio_list_devices in libsoundio**
Playback/recording through the PulseAudio sound server (default)
  id: default
  channel layouts:
    Mono
    Stereo
    2.1
    3.0
    3.0 (back)
    3.1
    4.0
    Quad
    Quad (side)
    4.1
    5.0 (back)
    5.0 (side)
    5.1
    5.1 (back)
    6.0 (side)
    6.0 (front)
    Hexagonal
    6.1
    6.1 (back)
    6.1 (front)
    7.0
    7.0 (front)
    7.1
    7.1 (wide)
    7.1 (wide) (back)
    Octagonal
  sample rates:
    1 - 192000
  formats: signed 24-bit LE, signed 24-bit BE, signed 32-bit LE, signed 32-bit BE, float 32-bit LE, float 32-bit BE
  min software latency: 0.00001563 sec
  max software latency: 0.68266667 sec

PulseAudio Sound Server
  id: pulse
  channel layouts:
    Mono
    Stereo
    2.1
    3.0
    3.0 (back)
    3.1
    4.0
    Quad
    Quad (side)
    4.1
    5.0 (back)
    5.0 (side)
    5.1
    5.1 (back)
    6.0 (side)
    6.0 (front)
    Hexagonal
    6.1
    6.1 (back)
    6.1 (front)
    7.0
    7.0 (front)
    7.1
    7.1 (wide)
    7.1 (wide) (back)
    Octagonal
  sample rates:
    1 - 192000
  formats: signed 24-bit LE, signed 24-bit BE, signed 32-bit LE, signed 32-bit BE, float 32-bit LE, float 32-bit BE
  min software latency: 0.00001563 sec
  max software latency: 0.68266667 sec

HDA Intel PCH, CX8200 Analog: Default Audio Device
  id: sysdefault:CARD=PCH
  probe error: unable to open device

HDA Intel PCH, CX8200 Analog: Direct sample mixing device
  id: dmix:CARD=PCH,DEV=0
  probe error: unable to open device

HDA Intel PCH, CX8200 Analog: Direct sample snooping device
  id: dsnoop:CARD=PCH,DEV=0
  probe error: unable to open device

HDA Intel PCH, CX8200 Analog: Direct hardware device without any conversions
  id: hw:CARD=PCH,DEV=0
  probe error: unable to open device

HDA Intel PCH, CX8200 Analog: Hardware device with all software conversions
  id: plughw:CARD=PCH,DEV=0
  probe error: unable to open device

Azure Kinect Microphone Array, USB Audio: Default Audio Device
  id: sysdefault:CARD=Array
  probe error: unable to open device

Azure Kinect Microphone Array, USB Audio: Direct sample mixing device
  id: dmix:CARD=Array,DEV=0
  probe error: unable to open device

Azure Kinect Microphone Array, USB Audio: Direct sample snooping device
  id: dsnoop:CARD=Array,DEV=0
  probe error: unable to open device

Azure Kinect Microphone Array, USB Audio: Direct hardware device without any conversions
  id: hw:CARD=Array,DEV=0
  probe error: unable to open device

Azure Kinect Microphone Array, USB Audio: Hardware device with all software conversions
  id: plughw:CARD=Array,DEV=0
  probe error: unable to open device

HDA Intel PCH CX8200 Analog (raw)
  id: hw:0,0
  probe error: unable to open device

Azure Kinect Microphone Array USB Audio (raw)
  id: hw:1,0
  probe error: unable to open device

**The output of libsoundio ./sio_microphone**
ALSA lib pcm_dmix.c:1052:(snd_pcm_dmix_open) unable to open slave
ALSA lib pcm_dsnoop.c:618:(snd_pcm_dsnoop_open) unable to open slave
ALSA lib pcm_dmix.c:1052:(snd_pcm_dmix_open) unable to open slave
ALSA lib pcm_dmix.c:990:(snd_pcm_dmix_open) The dmix plugin supports only playback stream
ALSA lib pcm_dsnoop.c:556:(snd_pcm_dsnoop_open) The dsnoop plugin supports only capture stream
ALSA lib pcm_dsnoop.c:618:(snd_pcm_dsnoop_open) unable to open slave
ALSA lib pcm_dsnoop.c:556:(snd_pcm_dsnoop_open) The dsnoop plugin supports only capture stream
ALSA lib pcm_dsnoop.c:556:(snd_pcm_dsnoop_open) The dsnoop plugin supports only capture stream
ALSA lib pcm_dsnoop.c:556:(snd_pcm_dsnoop_open) The dsnoop plugin supports only capture stream
ALSA lib pcm_dsnoop.c:556:(snd_pcm_dsnoop_open) The dsnoop plugin supports only capture stream
ALSA lib pcm_dsnoop.c:556:(snd_pcm_dsnoop_open) The dsnoop plugin supports only capture stream
ALSA lib pcm_dsnoop.c:618:(snd_pcm_dsnoop_open) unable to open slave
ALSA lib pcm_dmix.c:990:(snd_pcm_dmix_open) The dmix plugin supports only playback stream
ALSA lib pcm_dsnoop.c:618:(snd_pcm_dsnoop_open) unable to open slave
Input device: Playback/recording through the PulseAudio sound server
Output device: Playback/recording through the PulseAudio sound server

**Desktop info:**
 - OS with Version: Ubuntu 18.04.3 LTS
- version of kernel: 4.15.0-65-generic

**Hardware**
- output from lspci -v | grep -A7 -i "audio
 00:1f.3 Audio device: Intel Corporation Sunrise Point-LP HD Audio (rev 21) (prog-if 80)
	Subsystem: Hewlett-Packard Company Sunrise Point-LP HD Audio
	Flags: bus master, fast devsel, latency 64, IRQ 140
	Memory at 1fff018000 (64-bit, non-prefetchable) [size=16K]
	Memory at 1fff000000 (64-bit, non-prefetchable) [size=64K]
	Capabilities: <access denied>
	Kernel driver in use: snd_hda_intel
	Kernel modules: snd_hda_intel, snd_soc_skl

ETowleMS, 2019-12-10:
As Kinect DK has extremely sensitive microphones, you will find that the average signal appears low relative to full scale if you are reading the raw microphones. One reason for this is to avoid the microphone saturating in very loud conditions. 

For speech applications, we provide our Microsoft Audio Stack which cleans up the signal and applies relevant gain to give an appropriate output, see https://docs.microsoft.com/en-us/azure/cognitive-services/speech-service/speech-devices-sdk-windows-quickstart for the Kinect DK solution.

Thanks,
Erica

closed 2019-12-10 by tesych

busybeaver42, 2024-01-22:
If you are interessted into microphone array and kinect azure than have a look:
https://github.com/busybeaver42/kv3