microsoft/Azure-Kinect-Sensor-SDK #847: Missing k4a.pyd file

cbg88, 2019-10-18:
**Describe the bug**
 Getting linker errors, missing k4a.pyd file even after downloading the newest version (1.3). File cannot be open as it does not exist in the SDK.

**To Reproduce**

Should be a valid file path (if this file exists, however it does not within our machine). 
File should open. 

**Logs**

**Screenshots**

**Desktop (please complete the following information):**
 - OS with Version: 
 - SDK Version:  
 - Firmware version:  

**Additional context**

wes-b, 2019-10-19:
Sorry I can't tell what you are compiling. The SDK itself does not include Python.

closed 2019-10-24 by wes-b

lpasselin, 2019-10-31:
It is a typo  from the python sdk you are using.
.pyd files are not required btw. The instructions are wrong.

Try using this python sdk instead: https://github.com/etiennedub/pyk4a
Windows install instruction are provided.