microsoft/Azure-Kinect-Sensor-SDK #848: How should I pass the output.mkv which contains RGB+Depth to k4abt_simple_3d_viewer.exe ?

monajalal, 2019-10-18:
How should I pass the output.mkv which contains RGB+Depth to k4abt_simple_3d_viewer.exe ?

```
C:\Program Files\Azure Kinect Body Tracking SDK\tools>k4abt_simple_3d_viewer.exe -d WFOV_UNBINNED C:\Users\ivcla\workspace\kinectoutput\shreya_oct17.mkv

USAGE: (k4abt_)simple_3d_viewer.exe Mode[NFOV_UNBINNED, WFOV_BINNED]
    NFOV_UNBINNED (default) - Narraw Field of View Unbinned Mode [Resolution: 640x576; FOI: 75 degree x 65 degree]
    WFOV_BINNED             - Wide Field of View Binned Mode [Resolution: 512x512; FOI: 120 degree x 120 degree]
e.g.   (k4abt_)simple_3d_viewer.exe WFOV_BINNED

```

RoseFlunder, 2019-10-18:
This is not a feature of the simple 3d viewer.
You would have to integerate it yourself.
The source code for the simple 3d viewer is available in the samples:
https://github.com/microsoft/Azure-Kinect-Samples/tree/master/body-tracking-samples/simple_3d_viewer

monajalal, 2019-10-18:
well, I am requesting it as a feature then from Azure Kinect DK team

yijiew, 2019-10-21:
@monajalal : thanks for putting this request. Since it is not a bug, I will remove the "Bug" tag. 

As @RoseFlunder mentioned, the viewer is open-sourced. Please feel free to modify the code by yourself. We recently posted another sample that demonstrates how you playback a mkv clip and run through the body tracking SDK here: https://github.com/microsoft/Azure-Kinect-Samples/tree/master/body-tracking-samples/offline_processor. You can try to merge these two samples together.

RoseFlunder, 2019-10-21:
@yijiew
One question regarding the samples projects:
All projects are Visual Studio projects which is fine for Windows but would it be possible to provide CMake files or sth for Linux as well?

yijiew, 2019-10-21:
@RoseFlunder : we currently don't have the plan to do so. But it should be straightforward to create your own CMake files for Linux build. Since the sample project is open-sourced, if you create some CMake files that is working, we welcome you to send us a pull request :)

billpottle, 2020-04-03:
@monajalal - You can now use offline mode in simple_3d_viewer to view pre-recorded files.

qm13, 2020-04-25:
Fixed

closed 2020-04-25 by qm13