microsoft/Azure-Kinect-Sensor-SDK #849: How to calculate and utilize skeletal tracking and joint angles?

devLupin, 2019-10-21:
I want to use the skeletal tracking API that was supported in Kinect v2. There is no special documentation at this time, and it seems to be updated in the future.

It would be nice to include all the features of the existing Kinect v2.

My purpose is to calculate and utilize skeletal tracking and joint angles.

Currently I have developed in the following order:
Kinect v2 - Realsense (Intel) - Azure Kinect

In my opinion, realsense and azure kinect look very similar.
So it seems easier than developing for the first time.

The Realsense SDK did not have an API for the same purpose as Kinect v2. Also I'm a beginner developer and don't have the ability to implement such code.
So, I used a paid API called 'Nuitrack sdk' when using Realsense.

Please let me know if you have your own API for my purposes.
Thank you for Azure Kinect developer.

venscn, 2019-10-24:
You can see issue #844

venscn, 2019-10-24:
You can refer to the #844 issue link.

qm13, 2019-10-28:
You can use the [Azure Kinect Body Tracking SDK](https://docs.microsoft.com/en-us/azure/Kinect-dk/body-sdk-download.).

closed 2019-10-29 by qm13