microsoft/Azure-Kinect-Sensor-SDK #850: Ubuntu 18.04 - capture with multiple units on the same host machine not working

stbjelcevic, 2019-10-22:
I have a 2 Azure Kinect camera setup on Ubuntu 18.04 with an Nvidia 1080 GPU. I have been attempting to follow these step to record with both cameras at the same time using the external sync: https://docs.microsoft.com/en-us/azure/kinect-dk/record-external-synchronized-units#recording-when-multiple-units-connected-to-single-host-pc

I am able to successfully start the subordinate camera, but whenever I attempt to start the master camera, I get an error (shown below). In fact, any attempt to view/capture color + depth from 2 cameras at once is unsuccessful.

**To Reproduce**
1. Follow the instructions in the link on an Ubuntu 18.04 machine with 2 Kinect cameras connected.

**Expected behavior**
A .mkv file from both the cameras with equal frame counts.

**Logs**
This is the error I receive whenever I attempt to start capture from a second Kinect camera connected to the same computer as another Kinect that is already capturing:

k4arecorder --device 0 --external-sync master --imu OFF -e -8 -r 5 -l 5 output-0.mkv
Device serial number: 000873393812
Device version: Rel; C: 1.6.102; :anguished: 1.6.75[6109.7]; A: 1.6.14
libusb: error [submit_bulk_transfer] submiturb failed error -1 errno=12
[2019-10-16 10:45:53.241] [error] [t=26454] /home/vsts/work/1/s/extern/Azure-Kinect-Sensor-SDK/src/usbcommand/usbcommand.c (636): usb_cmd_io(). Error calling libusb_bulk_transfer for rx, result:LIBUSB_ERROR_IO
[2019-10-16 10:45:53.241] [error] [t=26454] /home/vsts/work/1/s/extern/Azure-Kinect-Sensor-SDK/src/usbcommand/usbcommand.c (739): usb_cmd_io(usbcmd_handle, cmd, p_cmd_data, cmd_data_size, p_data, data_size, NULL, 0, bytes_read, &cmd_status) returned failure in usb_cmd_read()
[2019-10-16 10:45:53.241] [error] [t=26454] /home/vsts/work/1/s/extern/Azure-Kinect-Sensor-SDK/src/depth_mcu/depth_mcu.c (389): usb_cmd_read( depthmcu->usb_cmd, DEV_CMD_NV_DATA_GET, (uint8_t *)&nv_tag, sizeof(nv_tag), calibration, cal_size, bytes_read) returned failure in depthmcu_get_cal()
[2019-10-16 10:45:53.241] [error] [t=26454] /home/vsts/work/1/s/extern/Azure-Kinect-Sensor-SDK/src/depth/depth.c (439): depthmcu_get_cal(depth->depthmcu, depth->calibration_memory, DEPTH_CALIBRATION_DATA_SIZE, &depth->calibration_memory_size) returned failure in depth_start()
[2019-10-16 10:45:54.241] [error] [t=26454] /home/vsts/work/1/s/extern/Azure-Kinect-Sensor-SDK/src/depth_mcu/depth_mcu.c (359): cmd_status == CMD_STATUS_PASS returned failure in depthmcu_depth_stop_streaming()
[2019-10-16 10:45:54.241] [error] [t=26454] /home/vsts/work/1/s/extern/Azure-Kinect-Sensor-SDK/src/depth_mcu/depth_mcu.c (362): depthmcu_depth_stop_streaming(). ERROR: cmd_status=0x00000063
[2019-10-16 10:45:54.242] [error] [t=26454] /home/vsts/work/1/s/extern/Azure-Kinect-Sensor-SDK/src/sdk/k4a.c (891): depth_start(device->depth, config) returned failure in k4a_device_start_cameras()
Runtime error: k4a_device_start_cameras(device, device_config) returned 1

**Desktop (please complete the following information):**
 - OS with Version: Ubuntu 18.04
 - SDK Version:  1.2 -> will try to update and see if problem persists.
 - Firmware version:  Latest

RoseFlunder, 2019-10-22:
If you don't have any problems with 1 kinect and only with 2 simultaneously than this might help:
https://github.com/microsoft/Azure-Kinect-Sensor-SDK/issues/485#issuecomment-510992581

By the way: Did you copy the k4a-rules files already?

stbjelcevic, 2019-10-24:
I did copy the rules file and can run the cameras without sudo, I will try the things mentioned in #485 and update.

closed 2019-10-24 by stbjelcevic