microsoft/Azure-Kinect-Sensor-SDK #855: ordered point cloud format

yuliangxiao, 2019-10-28:
How to export ordered point cloud format?Thanks

tesych, 2019-10-28:
@yuliangxiao, can you provide more information about what you need?

rabbitdaxi, 2019-10-29:
@yuliangxiao please look at whether this function provide what you need already:  [k4a_transformation_depth_image_to_point_cloud()](https://microsoft.github.io/Azure-Kinect-Sensor-SDK/master/group___functions_ga7385eb4beb9d8892e8a88cf4feb3be70.html#ga7385eb4beb9d8892e8a88cf4feb3be70). Please read the remarks in this function description that it output an xyz_image which is an organized point cloud. Here is an [example](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/b9463447fde0d9d1148c255acc49745f23f0db77/examples/transformation/main.cpp#L48) of how you can use the API

yuliangxiao, 2019-10-30:
Hello, this is the format I need。
[image]
I don't know how to get the last parameter(z),Thanks

yuliangxiao, 2019-10-31:
Thank you. I know what to do.

closed 2019-10-31 by yuliangxiao