microsoft/Azure-Kinect-Sensor-SDK #856: Unable to find a device on XUbuntu 18.04 LTS

AdrienPfeufferCarmenta, 2019-10-28:
**Describe the bug**
When trying to access my device on XUbuntu 18.04 LTS, k4aviewer and AzureKinectFirmwareTool both report that libusb devices are all unavailable. K4aviewer runs, but is unable to connect to a device, AzureKinectFirmwareTool only returns errors.

**To Reproduce**
1. Start with a fresh install of XUbuntu 18.04 LTS on a Lenovo Thinkpad T470p.
2. Install the 1.3 version of the Azure Kinect DK.
3. Update the device firmware on a separate Windows machine (it won't work on the Linux machine, as outlined below - and I also couldn't find the firmware blob anywhere in the Linux install).
4. Open a Terminal window.
5a. Type "AzureKinectFirmwareTool -q"
```
 == Azure Kinect DK Firmware Tool == 
[2019-10-28 09:11:36.038] [critical] [t=2786] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/usbcommand/usbcommand.c (360): find_libusb_device(). libusb device(s) are all unavalable. Is the device being used by another application?
[2019-10-28 09:11:36.038] [error] [t=2786] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/depth_mcu/depth_mcu.c (68): usb_cmd_create(USB_DEVICE_DEPTH_PROCESSOR, device_index, NULL, &depthmcu->usb_cmd) returned failure in depthmcu_create()
[2019-10-28 09:11:36.042] [critical] [t=2786] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/usbcommand/usbcommand.c (360): find_libusb_device(). libusb device(s) are all unavalable. Is the device being used by another application?
[2019-10-28 09:11:36.042] [error] [t=2786] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/color_mcu/color_mcu.c (74): usb_cmd_create(USB_DEVICE_COLOR_IMU_PROCESSOR, device_index, NULL, &colormcu->usb_cmd) returned failure in colormcu_create_by_index()
ERROR: Unable to find a device
```
5b. Type "k4aviewer"; In the upper left corner under "Open Device" the dropdown will list "(no available devices)". The terminal shows:
```
[2019-10-28 09:10:59.217] [critical] [t=2748] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/usbcommand/usbcommand.c (360): find_libusb_device(). libusb device(s) are all unavalable. Is the device being used by another application?
[2019-10-28 09:10:59.217] [error] [t=2748] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/depth_mcu/depth_mcu.c (68): usb_cmd_create(USB_DEVICE_DEPTH_PROCESSOR, device_index, NULL, &depthmcu->usb_cmd) returned failure in depthmcu_create()
[2019-10-28 09:10:59.217] [error] [t=2748] /__w/1/s/extern/Azure-Kinect-Sensor-SDK/src/sdk/k4a.c (133): depthmcu_create(index, &device->depthmcu) returned failure in k4a_device_open()
```

**Expected behavior**
5a. I get a list of the firmware and driver versions.
5b. I can open a device and view the video, IR and distance streams.

**Desktop (please complete the following information):**
 - OS with Version: XUbuntu 18.04 LTS
 - SDK Version:  1.3.0
 - Firmware version: 
RGB camera: 1.6.102
Depth camera: 1.6.75
Depth config: 6109.7
Audio: 1.6.14

RoseFlunder, 2019-10-28:
Did you copy the rules file as decribed here?
https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/usage.md#linux-device-setup

AdrienPfeufferCarmenta, 2019-10-28:
Yup, copied the file. Are the permissions correct?
`$ ll /etc/udev/rules.d/99-k4a.rules 
-rw-r--r-- 1 root root 771 Okt 28 12:08 /etc/udev/rules.d/99-k4a.rules`
(btw, just tried running k4aviewer as root, which works fine - so you're probably on the right track there)

rmbashirov, 2019-11-18:
Have worked with those azure kinect devices on 4 different computers, all running Ubuntu 18.04 and sdk version 1.3.0.

1. Had 3 different usb controllers: 
var 1) startech 4 driver usb controller works fine (not sure about amd);
var 2) i don't see devices in ./enumerate_devices;
var 3) i see devices in ./enumerate_devices but get usb errors when trying to record.
Had seen some controllers stop working when moving from intel to amd, can't recall exact configuration.

2. Mostly all usb ports from motherboards provide usb errors when trying to record. Behaviour: errors happening randomly when executing ./enumerate_devices.

3. Some PC power supplies do not provide enough power, so double check your additional current from power cable.

wes-b, 2019-12-05:
Closing, happy to reopen if this is still an issue.

closed 2019-12-05 by wes-b