microsoft/Azure-Kinect-Sensor-SDK #857: Unable to install libk4abt0.9(0.9.4) on ubuntu 18.04 LTS with k4a-tools(1.3.0).

SongJuHeon, 2019-10-29:
Hi, 
I installed  k4a-tools(1.3.0) on ubuntu 18.04 LTS. 
Then, I was trying to install  libk4abt0.9(0.9.4) and faced this issue below: 

[image]

Currently, I know the issue about installing both libk4a1.2 and libk4a1.3 at the same time. 
However, I really need both libk4a1.3 and libk4abt0.9(0.9.4) version. 

Is there any plan for supporting libk4abt0.9(0.9.4) on libk4a1.3 ?

Thanks.

OS: Ubuntu 18.04 LTS
SDK: Azure Kinect Sensor v1.3.0(k4atools v1.3.0, libk4a1.3, libk4a1.3-dev)

yijiew, 2019-11-05:
Thanks for reporting this issue. The problem will be solved in the future release. This issue is the same as #838 . I will close this one as duplication. Please use the other issue to monitor the fix.

yijiew, 2019-11-05:
#838

closed 2019-11-05 by yijiew