microsoft/Azure-Kinect-Sensor-SDK #858: The example "undistort"  always computes the xy range for depth camera

lihk11, 2019-10-29:
**Describe the bug**
 The example "undistort"  always compute the xy range for depth camera
Like the following line, 
https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/b9463447fde0d9d1148c255acc49745f23f0db77/examples/undistort/main.cpp#L131

When one wants to create an undistorted color image, or an undistorted depth image which aligned to the color image, the example will give a wrong result.

rabbitdaxi, 2019-10-29:
@lihk11  thank you for the feedback. Quote from [readme.md](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/examples/undistort/README.md) "The undistort example demonstrates how to undistort a depth map.". The initial idea of having this example is just trying to help people to understand how to undistort a depth map (given there are invalid pixels with depth value 0). Undistort a color image is well known by many open source libs such as opencv bilinear. However, we are happy to take your feedback to extend the example here to support both depth and color. We will mark this issue as enhancement.

lihk11, 2019-10-29:
You are right, but when it comes to depth map aligned to color image, the
function still could be confusing...

Even though, given that there have already been  a
k4a_calibration_type_t parameter
of create_pinhole_from_xy_range, the other function called by it should
using this parameter instead of a const value, which make the code more
beautiful.

rabbitdaxi, 2019-10-30:
@lihk11 agreed, added this to the enhancement list and will make a change for it.

PR #874 (2019-11-01): Helper fix in undistort example

## Fixes #858 

### Description of the changes:
- Simple fix of a helper in undistort to take camera type instead of just depth

### Before submitting a Pull Request:
- [x] I reviewed [CONTRIBUTING.md](https://github.com/Microsoft/Azure-Kinect-Sensor-SDK/blob/develop/CONTRIBUTING.md)
- [x] I [built my changes](https://github.com/Microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/building.md) locally
- [x] I ran the [unit tests](https://github.com/Microsoft/Azure-Kinect-Sensor-SDK/blob/develop/docs/testing.md)
- [x] I ran the [functional tests](https://github.c [... cut]
closed 2019-11-01 by rabbitdaxi