microsoft/Azure-Kinect-Sensor-SDK #859: Suppress Error and Event Logging with the C# wrapper

ahoink, 2019-10-29:
**Is your feature request related to a problem? Please describe.**
There's no way (as far as I can tell) to suppress error messages using the C# wrapper. Error messages print out even when the exception is caught in a try-catch.

**Describe the solution you'd like**
Expose the ability to silence error and event logging through the C# wrapper

**Describe alternatives you've considered**
Catching the exception (didn't work). Ignoring the issue (doesn't solve...).

**Additional context**
```
try
{
    kinect = Device.Open();
}
catch (AzureKinectOpenDeviceException)
{
     return false;
}
```

still results in

```
[2019-10-29 08:51:39.234] [critical] [t=13652] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\usbcommand\usbcommand.c (360): find_libusb_device(). libusb device(s) are all unavalable. Is the device being used by another application?

[2019-10-29 08:51:39.236] [error] [t=13652] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\depth_mcu\depth_mcu.c (68): usb_cmd_create(USB_DEVICE_DEPTH_PROCESSOR, device_index, NULL, &depthmcu->usb_cmd) returned failure in depthmcu_create()

[2019-10-29 08:51:39.236] [error] [t=13652] D:\a\1\s\extern\Azure-Kinect-Sensor-SDK\src\sdk\k4a.c (133): depthmcu_create(index, &device->depthmcu) returned failure in k4a_device_open()
```

MsDerekMa, 2019-10-29:
How are you setting the environment variable? I am seeing the error messages, but if I set the environment variable in the command line before launching my app it is suppressed. Are you attempting to suppress the logging from within in your program?

ahoink, 2019-10-29:
I set the environment variable via Control Panel > System > Advanced System Settings > Environment Variables. I couldn't find a way to suppress messages from my program.

MsDerekMa, 2019-10-30:
If you update the environment variable via the Advanced System Settings -> Environment Variables, you need to restart the process that you are launching your program from.  For example, restarting the Command Prompt or Visual Studio.

An easy way to check if it is set correctly is to add `Console.WriteLine("K4A_ENABLE_LOG_TO_STDOUT = {0}", Environment.GetEnvironmentVariable("K4A_ENABLE_LOG_TO_STDOUT"));` to your code before you make any K4A calls.  One note here is that calling `SetEnvironmentVariable` in C# won’t update the C runtime cache and therefore K4A won’t honor that setting.

Probably the easiest way to ensure that you don’t have the extra logging is to just run the executable from a batch script that first sets the environment variable before executing your program.

The original design was that by setting the debug message handling the console logging wouldn’t be initialized.  We are looking into fixing that.  However, because of the way the C# code was written, it always registers a handler, so fixing the issue would result in the opposite behavior you have now, if you want the logging you would have to set the environment variable to log to the console before launching the application.

Let me know if restarting the application lets you pick up the environment variable correctly.

ahoink, 2019-10-30:
It works now. Restarting Visual Studio fixed the issue. Thanks for the additional information.

closed 2019-10-30 by ahoink