microsoft/Azure-Kinect-Sensor-SDK #860: USB Set Address Failed (code 43)

lordjesus, 2019-10-29:
**Describe the bug**

The Azure Kinect camera fails to load correctly after the computer has been shut down and started again. When the computer is started again, the camera does not show up in the k4aviewer, and refreshing doesn't help. Looking in Device Manager, the 4K Camera shows up, but not the depth camera. Instead, there is a "Unknown USB Device (Set Address Failed)" (see screenshot below). Unplugging and re-plugging either of the cables fixes this. This doesn't happen when restarting the computer, only when shutting down. This is a problem because we use the camera with a computer inside a box that is not always connected to power. So right now we have to open the box and unplug and re-plug the cable every time we turn on our box.

I have tested this with three different cameras and three computers. The computers we use is [HP 260 G3 DM](https://support.hp.com/hr-en/document/c06098154) with i3-7130U CPU and Windows 10 (1903)

**To Reproduce**

1. Connect the camera to the computer, with both data and power cable in USB ports in the computer.
2. Verify that the camera is showing up correctly in k4aviewer and Device Manager.
3. Shut down the computer.
4. Turn on the computer.
5. Observe that the camera is not showing up in the k4aviewer and that it is showing as Unknown USB Device (Set Address Failed) in the Device Manager.
6. Unplug and re-plug one of the cables and see that it is now connected correctly.

**Expected behavior**

The camera should work when you turn on the computer. It shouldn't be necessary to unplug and re-plug the cables for the camera to work.

**Logs**

I tried to enable logging to file with environment variables, but I didn't understand how to do it. If any logs are needed, please tell me how I can enable the logging.

**Screenshots**

Screenshot of k4aviewer and Device Manager when the camera is correctly connected.

[image]

Screenshot of k4aviewer and Device Manager after a shutdown where the USB failed for the Depth Camera. Note that only the 4K Camera is shown, and there is a Unknown USB Device.

[image]

Screenshot of System Information from the computer.

[image]

**Desktop (please complete the following information):**
 - OS with Version: Windows 1903
 - SDK Version: 1.2 and 1.3.0
 - Firmware version:  1.6.102075014

**Additional context**

tesych, 2019-10-29:
@lordjesus , thank you for reporting the bug. We will address it with the firmware team.

lordjesus, 2019-12-20:
@tesych Any updates or ETA on this issue?

wes-b, 2019-12-20:
We are investigating a regression that turned up in testing. Our goal is to release in early January.

lordjesus, 2020-02-03:
@wes-b Any updates or upcoming releases?

wes-b, 2020-02-03:
Can you try the Alpha firmware in this package? https://www.nuget.org/packages/Microsoft.Azure.Kinect.Sensor/1.4.0-alpha.2

There are regressions in AzureKinectDK_Fw_1.6.107078014.bin with Linux, so we won't be promoting the firmware, but it has all USB fixes we are planning to make in the short term.

closed 2020-03-13 by wes-b