microsoft/Azure-Kinect-Sensor-SDK #861: Typo in k4abt_joint_id_t

hobinjk-ptc, 2019-10-29:
**Describe the bug**

The enum member K4ABT_JOINT_SPINE_NAVAL of k4abt_joint_id_t should be K4ABT_JOINT_SPINE_NAVEL since it's referring to the navel.

**To Reproduce**
1. Go to [k4abttypes.h](https://microsoft.github.io/Azure-Kinect-Body-Tracking/release/0.9.x/k4abttypes_8h_source.html)
2. Read members of k4abt_joint_id_t
3. See naval 🚢 

**Expected behavior**

Should be K4ABT_JOINT_SPINE_NAVEL

**Logs**
N/A

**Desktop (please complete the following information):**
 - OS with Version: N/A 
 - SDK Version:  Present in k4abt 0.9.4
 - Firmware version:  N/A

Zumra, 2019-11-05:
@hobinjk-ptc Thank you for reporting this. The issue will be fixed in the next release.

qm13, 2019-12-07:
Fixed in v0.9.5. Note that this is a breaking change. Joint K4ABT_JOINT_SPINE_NAVAL -> K4ABT_JOINT_SPINE_NAVEL.

closed 2019-12-07 by qm13