microsoft/Azure-Kinect-Sensor-SDK #862: multiple device sync: how to set up sync-delay

iammarvelous, 2019-10-30:
**Describe the bug**

Hi, I was going through documents about multiple devices sync. I finally ended up here because I was wondering how to set sync-delay. I think [this document](https://docs.microsoft.com/en-us/azure/Kinect-dk/record-external-synchronized-units) should also be updated to avoid confusion. To be clear, I want to check with you that the following is the correct way to do multiple devices sync:
```
# for master
k4arecorder.exe --device 0 --external-sync master output-1.mkv
# for subordinate-1
k4arecorder.exe --device 1 --external-sync subordinate --sync-delay 0.16 output-2.mkv
# for subordinate-2
k4arecorder.exe --device 2 --external-sync subordinate --sync-delay 0.32 output-3.mkv
```

**To Reproduce**

**Expected behavior**

**Logs**

**Screenshots**

**Desktop (please complete the following information):**
 - OS with Version: 
 - SDK Version:  
 - Firmware version:  

**Additional context**

wes-b, 2019-10-30:
Thanks, we will get that page updated.

For multi sync please review #530. 
To set the sync-delay you choose the delay in micro seconds. The resolution is 11us, so anything smaller than that will be lost.
You should also start all the subordinates first and master last. 

Here is a script I have used in the past for testing:
```
start "sub 1"  cmd /c k4arecorder.exe --device 2 -l 10 --imu off -r 5 -e -6 -d WFOV_UNBINNED -c 1080p --external-sync Subordinate sub1.mkv ^> sub1.mkv.txt
ping 127.0.0.1 -n 2 > null
start "sub 2"  cmd /c k4arecorder.exe --device 1 -l 10 --imu off -r 5 -e -6 -d WFOV_UNBINNED -c 1080p --external-sync Subordinate sub2.mkv ^> sub2.mkv.txt
ping 127.0.0.1 -n 3 > null
start "master" cmd /c k4arecorder.exe --device 0 -l 10 --imu off -r 5 -e -6 -d WFOV_UNBINNED -c 1080p --external-sync Master master.mkv ^> master.mkv.txt
```

iammarvelous, 2019-10-30:
It's strange. I don't see you pass anything into sync-delay in the above scripts. Is it intentional or a typo?