microsoft/Azure-Kinect-Sensor-SDK #863: k4a_calibration_2d_to_3d

yuliangxiao, 2019-10-30:
Why there is no Z coordinate?thanks
[image]

wes-b, 2019-10-30:
I don't understand your question. Please elaborate more.

yuliangxiao, 2019-10-31:
Parameter is "1.f",I don't know the Z coordinate has always been 1.000000,The document says target_point3d_mm unit is mm,Why is the parameter value is -0.1388,Instead of 1388,Where can I find such documents?thanks

venscn, 2019-10-31:
In the last step of function k4a_calibration_2d_to_3d, x = target.x * source_depth_mm,y = target.y * source_depth_mm, z = source_depth_mm.

yuliangxiao, 2019-10-31:
I know that,But I want to know why XY is -0.1388,Unit is also mm?thanks

rabbitdaxi, 2019-10-31:
@yuliangxiao please read through this page first: https://docs.microsoft.com/en-us/azure/Kinect-dk/use-calibration-functions, and please also read the function doc remarks and parameter descriptions. Make sure when you using this function, you need to check the function return valid result. Let us know whether you still have questions after you read the link above. It might be helpful if you describe what exactly trying to do, you can paste sample code if you want us to help you spot your issue

closed 2019-10-31 by yuliangxiao