microsoft/Azure-Kinect-Sensor-SDK #864: simple_3d_viewer does not have CMake file

legacydev, 2019-10-30:
The linux installation has the simple_3d_viewer demo and can run it on Ubuntu with no issue. However, the source does not include the CMake file. Can you please include the CMake file in the source?

legacydev, 2019-10-30:
Could we also get the offline_processor CMake file?

wes-b, 2019-10-30:
> Could we also get the offline_processor CMake file?

Can you share a link to what you are referring to?

legacydev, 2019-10-30:
https://github.com/microsoft/Azure-Kinect-Samples/tree/master/body-tracking-samples/offline_processor
and 
https://github.com/microsoft/Azure-Kinect-Samples/tree/master/body-tracking-samples/simple_3d_viewer

wes-b, 2019-10-31:
https://github.com/microsoft/Azure-Kinect-Samples/pull/34

wes-b, 2019-11-05:
I was not able to add offline_processor, but there should be enough in place now to get the other projects building with CMake.

closed 2019-11-05 by wes-b

legacydev, 2019-11-07:
Thanks for the quick response! I haven't been able to build this successfully, yet. I've attached the output I am receiving. I can build the main sdk and examples
[Output-CMake.txt](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/files/3820948/Output-CMake.txt)

wes-b, 2019-11-07:
Ok, there is some build information in the readme.md. If cloning the repo and installing the k4a packages doesn't work for you then please open a new bug and use our template to provide repro steps and other helpful information.

https://github.com/microsoft/Azure-Kinect-Samples/blob/master/README.md#linux

legacydev, 2019-11-07:
As I am building with Visual Studio on Windows, I am unable to run the following commands:

apt install libk4abt0.9-dev
apt install libxi-dev

When attempting to install via vcpkg throws an error "Contains invalid characters. Only alphanumeric lowercase ASCII characters and dashes are allowed"
The cmake build prompted me to install dependencies for the main Kinect DK. I expect the same for these example projects, but this doesn't seem to be the case. Do you want me to open a new issue for that?

legacydev, 2019-11-07:
Actually, I am unable to install libk4abt0.9-dev on Ubuntu either. Is the command correct?