microsoft/Azure-Kinect-Sensor-SDK #865: How to minimize depth -> color frame alignment inaccuracy

amonnphillip, 2019-10-30:
I am wondering what techniques I can use to eliminate the depth to color frame alignment issues seen even in the azure dk viewer tool (see the attached image). The image shows color bleed on a part of the point cloud that should be black, and parts of the reference board texture is shown on background object..

In my own project I am using the intrinsic and extrinsic values to select a pixel from the color frame that most accurately represents the depth pixel using the code in k4a_calibration_3d_to_3d, then k4a_calibration_3d_to_2d and I see a similar issue. 

Is the effect seen in the image because is inaccurate calibration? And thus there isn't much I can do?
 
Or is it due to the algorithm I use to map a 3d point in world space to the color frame?

Or is is a fundamental hardware issue due to the depth sensor for example?

[image]

amonnphillip, 2019-11-01:
Looking at this further it looks like occlusion, where the depth camera and the color camera do not see the same regions, so a miss-mapping of pixels in the color frame are projected on to the point cloud.

rabbitdaxi, 2019-11-04:
@amonnphillip thank you for the feedback. Hope my comments below can help:
- For those k4aviewer images you pasted above, internally, k4aviewer already uses a function that tried to dealing with occlusion [k4a_transformation_depth_image_to_color_camera()](https://microsoft.github.io/Azure-Kinect-Sensor-SDK/master/group___functions_gafacffb5f781a9c2df30d4a16241cd514.html#gafacffb5f781a9c2df30d4a16241cd514). You can read this [page ](https://docs.microsoft.com/en-us/azure/Kinect-dk/use-image-transformation) for more details. And the reason you still see some of these alignment error in k4aviewer more likely due to that given depth and color have different resolution (depth normally is less resolution than color), even slightly depth edge imperfection from depth camera can be enlarged during the transformation. 
- If you are using [k4a_calibration*](https://docs.microsoft.com/en-us/azure/Kinect-dk/use-calibration-functions) functions, because these functions themselves are operating on single point, occlusion logic cannot be applied for these functions. However, if user is invoking these functions to transform more than 1 point and would like to z-check occlusion, user can do that similarly as what we do for the k4a_transformation_depth_image_to_color_camera(), which [reference code](https://github.com/microsoft/Azure-Kinect-Sensor-SDK/blob/develop/src/transformation/rgbz.c) is published in the SDK

closed 2019-11-07 by amonnphillip